nanpy/nanpy #39: Request: epir library and pulsein

ryanvade, 2015-01-24:
Any chance that the ePIR library and puslein functions could be added to nanpy?

ponty, 2015-01-26:
Do you have a link to the lib?

ryanvade, 2015-01-28:
https://code.google.com/p/epir-library-v1/

ponty, 2015-01-28:
The license is not compatible with Nanpy-firmware.
Nanpy-firmware should change its license to make this possible.

ryanvade, 2015-01-29:
Nanpy is not using GPL v3?

astagi, 2015-01-30:
Nanpy is released under MIT license, but afaik MIT is a GPL-compatible license, so it's possible to combine it with epir. Would you like to contribute to Nanpy making the porting by yourself @ryanvade ? here you can find some instructions http://nanpy.github.io/#how-to-extend-nanpy

ryanvade, 2015-01-30:
I got lost when following that page, at the part where you add the lib to the Arduino source.  Why is the QuickArray (or whatever its called) not just called list ? :)

I think I have the python and some of the c code ready though.

astagi, 2015-01-31:
Sounds great! test your code and open a pull request!

I'm sorry you got lost, if something is not clear for you please open a issue on documentation, I've just made a dedicated label for it!

For using "list" instead of SlimArray I don't remember.. Anyway it's just a name and it means that is a tinier version of the std lib array implementation.

ponty, 2015-02-02:
MIT can be combined with GPL, but the result should be GPL as I understand.

http://www.gnu.org/licenses/gpl-faq.html#IfLibraryIsGPL

"If a library is released under the GPL (not the LGPL), does that mean that any software which uses it has to be under the GPL or a GPL-compatible license? 
Yes, because the software as it is actually run includes the library."

astagi, 2015-02-03:
Uhm, he asks "GPL or GPL-compatible" license, so the software that uses it (Nanpy in this case uses epir lib) has to be under a GPL-compatible license (MIT license in this case). We can include epir library without any changes on license.

ponty, 2015-02-03:
OK!

astagi, 2015-02-11:
@ryanvade any updates here?

ryanvade, 2015-03-29:
I am attempting to add the PulseIn method from the standard library as follows:
```C++
//Robotics.cpp
#include "cfg.h"

#if USE_ROBOTICS

#include <Arduino.h>
#include "pulseIn.h"
#include "RoboticsClass.h"
#include <stdlib.h>

const char* nanpy::RoboticsClass::get_firmware_id()
{
    return "Robotics";
}

double nanpy::RoboticsClass::pulseIn( MethodDescriptor* m ) {
    ObjectsManager<Robotics>::pulseIn(m);

    if (strcmp(m->getName(),"new") == 0) {
        Robotics* robot;
        robot = new Robotics (m->getInt(0), m->getInt(1), m->getInt(2));
        robot->begin();
        v.insert(robot);
        m->returns(v.getLastIndex());
    }

    if (strcmp(m->getName(), "pulseIn") == 0) {
        m->returns(v[m->getObjectId()]->portWrite(m->getInt(0), m->getInt(1), m->getInt(2));
    }
};

#endif
```

```C++
//RoboticsClass.h
#ifndef ROBOTICS_CLASS
#define ROBOTICS_CLASS

#include "BaseClass.h"
#include "MethodDescriptor.h"

class Robotics;

namespace nanpy {
    class RoboticsClass: public ObjectsManager<Robotics> {
        public:
            double pulseIn( nanpy::MethodDescriptor* m );
            const char* get_firmware_id();
    };
}

#endif
```C++
//pulseIn.h
#include "wiring_private.h"
#include "pins_arduino.h"

class PulseIn;

using namespace std;

class PulseIn {
	public:
	unsigned long pulseIn(uint8_t pin, uint8_t state, unsigned long timeout)
{
	// cache the port and bit of the pin in order to speed up the
	// pulse width measuring loop and achieve finer resolution.  calling
	// digitalRead() instead yields much coarser resolution.
	uint8_t bit = digitalPinToBitMask(pin);
	uint8_t port = digitalPinToPort(pin);
	uint8_t stateMask = (state ? bit : 0);
	unsigned long width = 0; // keep initialization out of time critical area
	
	// convert the timeout from microseconds to a number of times through
	// the initial loop; it takes 16 clock cycles per iteration.
	unsigned long numloops = 0;
	unsigned long maxloops = microsecondsToClockCycles(timeout) / 16;
	
	// wait for any previous pulse to end
	while ((*portInputRegister(port) & bit) == stateMask)
		if (numloops++ == maxloops)
			return 0;
	
	// wait for the pulse to start
	while ((*portInputRegister(port) & bit) != stateMask)
		if (numloops++ == maxloops)
			return 0;
	
	// wait for the pulse to stop
	while ((*portInputRegister(port) & bit) == stateMask) {
		if (numloops++ == maxloops)
			return 0;
		width++;
	}

	// convert the reading to microseconds. The loop has been determined
	// to be 20 clock cycles long and have about 16 clocks between the edge
	// and the start of the loop. There will be some error introduced by
	// the interrupt handlers.
	return clockCyclesToMicroseconds(width * 21 + 16); 
}
};
```

IS this correct?

ryanvade, 2015-04-18:
No comment?

ponty, 2015-04-20:
1. The first file should be `RoboticsClass.cpp`

2. You should edit Nanpy.ino. Example:
```c
#if USE_DallasTemperature
#include <DallasTemperature.h>
#endif
//....
```

ryanvade, 2015-04-22:
I renamed the file and had already modified Nanpy.ino.  Did I do anything else wrong?

ponty, 2015-04-22:
Push a branch to your repo, so we can check it!

What is Robotics class?
I don't find it in epir library.

ryanvade, 2015-05-07:
Sorry this has turned into more of a how to enhance nanpy thread.  EPIR has been put on hold for the time being so I am trying to add pulsein support.

ponty, 2017-04-14:
PulseIn was added.

closed 2017-04-14 by ponty