nanpy/nanpy #42: Incorrect measurements with dallastemperature.py/mytemperaturelib.py

lunasilvestre, 2015-02-01:
Having some trouble to get correct temperature readings using DS18B20 with nanpy.
Tested the temperature probe from DFRobot's Analog EC Meter and, although it works fine with Arduino IDE based code - dallas and onewire examples -, couldn't get any reasonable readings using nanpy. E.g.: dallastemperature.py gives me -127C and mytemperaturelib.py gives 68C for a temperature around 8C.

After some debugging was able to get correct values with mytemperature.lib changing line 53 (?) from 'raw = raw << 3' to 'raw = raw & ~7'.

Pi3R1k, 2015-02-04:
-127 means no value. 
There seems to be a problem with address in dallastemperature.py exemple.

The address needs to be an array. Just add this just after  `addr = sensors.getAddress(0)` :
```
addr = [int(i) for i in addr.split()]
```

astagi, 2015-02-11:
Thanks @Pi3R1k , @lunasilvestre can you share some python code?

I'll deeply test dallas temperature before 0.9.3 release (next week), let you know.

astagi, 2015-02-14:
Tested with this code (here I'm not using 1 wire bus):

    from nanpy import DallasTemperature
    from nanpy import SerialManager

    #remember that you need to change serial device or let Nanpy autodetect it for you
    connection = SerialManager(device='/dev/cu.usbmodem141131')

    sensor1 = DallasTemperature(4, connection=connection)
    sensor2 = DallasTemperature(5, connection=connection)

    print("There are %d devices connected on pin %d" % (sensor1.getDeviceCount(), sensor1.pin))
    print("There are %d devices connected on pin %d" % (sensor2.getDeviceCount(), sensor2.pin))
    addr1 = sensor1.getAddress(0)
    addr2 = sensor2.getAddress(0)
    sensor1.setResolution(9)
    sensor2.setResolution(12)
    while True:
        sensor1.requestTemperatures(0)
        sensor2.requestTemperatures(0)
        temp1 = sensor1.getTempC(addr1)
        temp2 = sensor2.getTempC(addr2)
        print("The temperature 1, in Celsius degrees is %0.2f" % temp1)
        print("The temperature 2, in Celsius degrees is %0.2f" % temp2)

And here's a photo of my circuit (red for vcc, black for gnd and green for data)

[image]

closed 2015-02-14 by astagi

astagi, 2015-02-14:
Looks like getTempC/getTempF doesn't work with addresses! I get the same value for both connected devices. I suggest to use index and checkout the current release branch https://github.com/nanpy/nanpy/tree/feature/release0.9.3

Here's some code 

    from nanpy import DallasTemperature

    sensors = DallasTemperature(5)
    n_sensors = sensors.getDeviceCount()

    print("There are %d devices connected on pin %d" % (n_sensors, sensors.pin))
    addresses = []

    for i in range(n_sensors):
        addresses.append(sensors.getAddress(i))

    sensors.setResolution(12)

    while True:
        sensors.requestTemperatures()
        for i in range(n_sensors):
            temp = sensors.getTempC(i)
            print("Device %d (%s) temperature, in Celsius degrees is %0.2f" % (i, addresses[i], temp))
            print("Let's convert it in Fahrenheit degrees: %0.2f" % DallasTemperature.toFahrenheit(temp))
        print("\n")