nanpy/nanpy #45: Install failure on Python3 with pyzo

jjfPCSI1, 2015-02-14:
Hello,

I'm trying to install nanpy from within pyzo using the pip command but it fails with the given error message:

<pre>
In [1]: pip install nanpy
Downloading/unpacking nanpy
  Downloading nanpy-v0.9.2.tar.gz
  Running setup.py (path:/private/var/folders/20/_tnbfqpd59xcv116l3_c319r0000gn/T/pip_build_mylogin/nanpy/setup.py) egg_info for package nanpy
    Traceback (most recent call last):
      File "<string>", line 17, in <module>
      File "/Users/mylogin/Desktop/pyzo2014a/lib/python3.4/tokenize.py", line 437, in open
        buffer = builtins.open(filename, 'rb')
    FileNotFoundError: [Errno 2] No such file or directory: '/private/var/folders/20/_tnbfqpd59xcv116l3_c319r0000gn/T/pip_build_mylogin/nanpy/setup.py'
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):

  File "<string>", line 17, in <module>

  File "/Users/mylogin/Desktop/pyzo2014a/lib/python3.4/tokenize.py", line 437, in open

    buffer = builtins.open(filename, 'rb')

FileNotFoundError: [Errno 2] No such file or directory: '/private/var/folders/20/_tnbfqpd59xcv116l3_c319r0000gn/T/pip_build_mylogin/nanpy/setup.py'

----------------------------------------
Cleaning up...
Command python setup.py egg_info failed with error code 1 in /private/var/folders/20/_tnbfqpd59xcv116l3_c319r0000gn/T/pip_build_mylogin/nanpy
Storing debug log for failure in /Users/mylogin/.pip/pip.log
Exception information:
Traceback (most recent call last):
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.3/lib/python3.3/site-packages/pip/basecommand.py", line 122, in main
    status = self.run(options, args)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.3/lib/python3.3/site-packages/pip/commands/install.py", line 278, in run
    requirement_set.prepare_files(finder, force_root_egg_info=self.bundle, bundle=self.bundle)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.3/lib/python3.3/site-packages/pip/req.py", line 1229, in prepare_files
    req_to_install.run_egg_info()
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.3/lib/python3.3/site-packages/pip/req.py", line 325, in run_egg_info
    command_desc='python setup.py egg_info')
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.3/lib/python3.3/site-packages/pip/util.py", line 697, in call_subprocess
    % (command_desc, proc.returncode, cwd))
pip.exceptions.InstallationError: Command python setup.py egg_info failed with error code 1 in /private/var/folders/20/_tnbfqpd59xcv116l3_c319r0000gn/T/pip_build_mylogin/nanpy
 
</pre>

Trying directly from the command line gives the same problem.
Any idea on how it could be solved ?

Thanks

JJ

astagi, 2015-02-15:
Hello @jjfPCSI1 ,

sorry I don't know pyzo, what is it?

Please can you try installing Nanpy manually? Just download the zip (https://github.com/nanpy/nanpy/archive/master.zip) and launch 

    pip install .

Let me know :)

jjfPCSI1, 2015-02-15:
Hello @astagi ,

Pyzo is a portable distribution of Python (see pyzo.org) which is perfectly suited for students that don't have time to dive into installation problems and other things like that.

Thanks for the tip, it worked. There are still some errors due to python2 print syntax (Pyzo comes with Python3) but pip seems to be happy with the installation nonetheless :

<pre>
In [5]: pip install .
Unpacking /Users/monlogin/Downloads/nanpy-master
  Running setup.py (path:/var/folders/20/_tnbfqpd59xcv116l3_c319r0000gn/T/pip-fu907yb3-build/setup.py) egg_info for package from file:///Users/monlogin/Downloads/nanpy-master
    
Requirement already satisfied (use --upgrade to upgrade): pyserial in /Users/monlogin/Desktop/pyzo2014a/lib/python3.4/site-packages (from nanpy==0.9.2)
Installing collected packages: nanpy
  Running setup.py install for nanpy
      File "/Users/monlogin/Desktop/pyzo2014a/lib/python3.4/site-packages/nanpy/examples/i2c_scanner.py", line 15
        print connection.device
                       ^
    SyntaxError: invalid syntax
    
      File "/Users/monlogin/Desktop/pyzo2014a/lib/python3.4/site-packages/nanpy/examples/i2c_send.py", line 15
        print connection.device
                       ^
    SyntaxError: invalid syntax
    
      File "/Users/monlogin/Desktop/pyzo2014a/lib/python3.4/site-packages/nanpy/examples/pressure.py", line 20
        print 'pressure: %s kPa' % (p / 1000)
                               ^
    SyntaxError: invalid syntax
    
    
Successfully installed nanpy
Cleaning up...
In [5]: import nanpy

In [6]: 
</pre>

closed 2015-02-15 by jjfPCSI1

astagi, 2015-02-15:
No problem! :) I'll fix examples in the next release, thanks!

Also I'd like to try Pyzo and see what's the problem with Nanpy..