nanpy/nanpy #47: Objects

astagi, 2015-02-15:
Hi @ponty ,

unfortunately yesterday I noticed that class methods call the global serial manager instead of the serial manager tied with the object. This is absolutely normal, because they're class methods, but it's obviously a bug.

I really like the connection parameter, it allows people to use more than a board in the same script, but I think that we should change all the classes and transform class methods in object methods, something like you did with EEPROM (we have eepromobj.py, I think we should use only eepromobj.py in place of eeprom.py)

ponty, 2015-02-16:
`@arduinoclassmethod` should be replaced with `@arduinomethod`

First merge all your commits into master branch, then I start working on it.
I don't want to handle merge conflicts.

astagi, 2015-02-16:
Thanks! merged https://github.com/nanpy/nanpy/pull/46

astagi, 2015-02-16:
@ponty we have to decide if removing old EEPROM and switch to EEPROM object (still using module eeprom). No problem for me to remove the old EEPROM, it will break some old code but I think that having similar modules doing the same thing is confusing..

ponty, 2015-02-17:
I started a new test subsytem `MockConnection` before changing a lot of code to increase test coverage.
 `MockConnection`  simulates the firmware in tests.

ponty, 2015-02-17:
I just removed arduino.py and eeprom.py :)

astagi, 2015-02-17:
Nice :) I think that we should rename ArduinoApi in Arduino to maintain old compatibility, or define a new class Arduino that extends ArduinoApi

astagi, 2015-02-17:
+1 for MockConnection

ponty, 2015-02-18:
You can rename any class you want like

ArduinoApi -> Arduino

EepromObj -> EEPROM

astagi, 2015-02-18:
Ok :) for Arduino I'd let ArduinoApi, we need a fresh change also we used ArduinoAPi in latest README and examples..

enoac, 2015-04-24:
So i'm currently trying to use nanpy with my arduino mega 2560,  I have the firmware and everything loaded up.  However when I try to run some python code I just get a serial timeout.  Any suggestions?

astagi, 2015-04-27:
Hey @enoac ,

I experienced the same problem with Raspbian yesterday, I found this workaround: https://github.com/nanpy/nanpy/commit/a272ed9b20a362fda86cb2cfb1a069bd0a58a6de

Please try installing Nanpy from source

git clone git@github.com:nanpy/nanpy.git
cd nanpy
python setup.py install

and try using rtscts parameter in SerialManager. You can find an example here:

https://github.com/DjangoBeer/minecraftpi-hack/blob/master/altimeter/altimeter.py

Let me know if it works, this fix will be included in the next release, Nanpy 0.9.4.

closed 2017-04-14 by ponty