nanpy/nanpy #48: destroying objects

ponty, 2015-02-17:
@astagi 
Is there any solution for destroying objects? (problem was found in #29)

I propose to remove Python `__del__()` and C++ `SlimArray::remove()` from the code.
Do you have any use case where it is needed?

We can add a new command `clean` which removes all existing objects. This can be called at start of the script and the end explicitly.

astagi, 2015-02-17:
No solution atm, users must explicitly call "del". I'll try to think about it, I was thinking about a global register for all the instances and call "del" on them at the end of the script, or whenever the user wants.

ponty, 2015-02-18:
I accidentally commented out `__del__()`

`del x` doesn't call `x.__del__()` so the current solution is wrong. (read this http://stackoverflow.com/questions/6104535/i-dont-understand-this-python-del-behaviour) For destroying remote objects an explicit method should be called like `x.destroy()`

astagi, 2015-02-18:
ok let's go for a custom method :+1:

astagi, 2015-02-18:
To keep things simple I'd avoid to maintain a global register of instances and similar solutions, let the user clean his objects explicitly.