nanpy/nanpy #54: Problem with DallasTemperature and digitalWrite

xaratt, 2015-03-14:
Hi,

I'm trying to write simple thermostat, but found that after initializing DallasTemperature instance I can't write data into pin. I'm using RPi with Arduino UNO.

Code:
```python
from nanpy import SerialManager, ArduinoApi, DallasTemperature
import time
import logging
logging.basicConfig(level=logging.DEBUG)

def main():
    connection = SerialManager(device="/dev/ttyACM0", baudrate=115200)
    a = ArduinoApi(connection=connection)
    a.pinMode(13, a.OUTPUT)

    sensors = DallasTemperature(2, connection)  # (*)
    # setResolution & requestTemperatures here

    while True:
        a.digitalWrite(13, a.HIGH)
        time.sleep(1)
        a.digitalWrite(13, a.LOW)
        time.sleep(1)

if __name__ == "__main__":
    main()
```

and last lines of the logs:
```
DEBUG:nanpy.serialmanager:sending:'DallasTemperature\x00'
DEBUG:nanpy.serialmanager:sending:'0\x00'
DEBUG:nanpy.serialmanager:sending:'1\x00'
DEBUG:nanpy.serialmanager:sending:'new\x00'
DEBUG:nanpy.serialmanager:sending:'2\x00'
DEBUG:nanpy.serialmanager:received:'0\r\n'
DEBUG:nanpy.serialmanager:sending:'A\x00'
DEBUG:nanpy.serialmanager:sending:'0\x00'
DEBUG:nanpy.serialmanager:sending:'2\x00'
DEBUG:nanpy.serialmanager:sending:'dw\x00'
DEBUG:nanpy.serialmanager:sending:'13\x00'
DEBUG:nanpy.serialmanager:sending:'1\x00'
^C
```

Afrter removing line (*) all works fine, diod blinks.

astagi, 2015-03-20:
Have you edited cfg.h file including Dallas temperature library?

astagi, 2015-04-27:
Hey @xaratt ,

I experienced the same problem with Raspbian yesterday, I found this workaround: https://github.com/nanpy/nanpy/commit/a272ed9b20a362fda86cb2cfb1a069bd0a58a6de

Please try installing Nanpy from source

git clone git@github.com:nanpy/nanpy.git
cd nanpy
python setup.py install

and try using rtscts parameter in SerialManager. You can find an example here:

https://github.com/DjangoBeer/minecraftpi-hack/blob/master/altimeter/altimeter.py

Let me know if it works, this fix will be included in the next release, Nanpy 0.9.4.

astagi, 2016-02-06:
Too old... closing

closed 2016-02-06 by astagi