nanpy/nanpy #58: Support for Adafruit-PWM-Servo-Driver-Library

balzreber, 2015-08-06:
I'm trying to add support for the Adafruit-PWM-Servo-Driver-Library (https://learn.adafruit.com/16-channel-pwm-servo-driver/overview)
Unfortunately I'm not as successful as I hoped.
I have experience in web programming and java but not so much in python or c.
I suspect the problem is, that the Adafruit-PWM-Servo-Driver-Library is statically initialized, and not with the normal "new", but it could very well be something totally different...
Any help is much appreciated :-)

##### Error Message
> PWMServoClass.cpp: In member function 'virtual void nanpy::PWMServoClass::elaborate(nanpy::MethodDescriptor*)':
PWMServoClass.cpp:21:21: error: no matching function for call to 'nanpy::SlimArray<PWMServo*>::insert(Adafruit_PWMServoDriver&)'
         v.insert(pwm);
                     ^
PWMServoClass.cpp:21:21: note: candidates are:
In file included from BaseClass.h:5:0,
                 from PWMServoClass.h:4,
                 from PWMServoClass.cpp:7:
SlimArray.h:29:18: note: void nanpy::SlimArray<T>::insert(T) [with T = PWMServo*]
             void insert(T el) {
                  ^
SlimArray.h:29:18: note:   no known conversion for argument 1 from 'Adafruit_PWMServoDriver' to 'PWMServo*'
SlimArray.h:54:18: note: void nanpy::SlimArray<T>::insert(int, T) [with T = PWMServo*]
             void insert(int pos, T el) {
                  ^
SlimArray.h:54:18: note:   candidate expects 2 arguments, 1 provided

##### PWMServoClass.h
```
#ifndef PWMSERVO_CLASS
#define PWMSERVO_CLASS

#include "BaseClass.h"
#include "MethodDescriptor.h"

class PWMServo;

namespace nanpy {
    class PWMServoClass: public ObjectsManager<PWMServo> {
        public:
            void elaborate( nanpy::MethodDescriptor* m );
            const char* get_firmware_id();
    };
}

#endif
```

##### PWMServoClass.cpp
```
#include "cfg.h"

#if USE_PWMServo

#include <Arduino.h>
#include <Adafruit_PWMServoDriver.h>
#include "PWMServoClass.h"
#include <stdlib.h>

const char* nanpy::PWMServoClass::get_firmware_id()
{
    return "PWMServoDriver";
}

void nanpy::PWMServoClass::elaborate( MethodDescriptor* m ) {
    ObjectsManager<PWMServo>::elaborate(m);

    if (strcmp(m->getName(),"new") == 0) {
        Adafruit_PWMServoDriver pwm = Adafruit_PWMServoDriver();
        pwm1.begin();
        v.insert(pwm);
        m->returns(v.getLastIndex());
    }

    if (strcmp(m->getName(), "setPWMFreq") == 0) {
        //m->returns(v[m->getObjectId()].setPWMFreq(m->getFloat(0)));
    }

    if (strcmp(m->getName(), "setPWM") == 0) {
        //m->returns(v[m->getObjectId()].portRead(m->getInt(0), m->getInt(1), m->getInt(2)));
    }
};

#endif
```

astagi, 2015-08-08:
Hi @balzreber did you set to 1 USE_PWMServo in cfg.h? Did you include Adafruit library in the default libraries? Which version of firmware are you using? I suggest to download the latest firmware from pypi

balzreber, 2015-08-10:
Hi @astagi thank's for your quick answer.
Jep, in cfg.h file I have
```
#define USE_PWMServo                                1
```
further, here are all related code lines in Nanpy file (witch I don't included in the previous post).
```
#if USE_PWMServo
#include <Adafruit_PWMServoDriver.h>
#endif
```
```
#include "PWMServoClass.h"
```
```
REGISTER_CLASS_CONDITIONAL(PWMServoClass, USE_PWMServo);
```
I'm not sure witch firmware I was using.
I downloaded the latest Version (9.4) from pypi as you suggested, but still, I get the same Error message...
Do you think the rest of the code, as you see it in my last post, is correct???

Thanks a lot, for your help :-)

balzreber, 2015-08-11:
I tried with another Adafruit Product. The TLC5947 LED Controller (https://learn.adafruit.com/tlc5947-tlc59711-pwm-led-driver-breakout/overview).
But I get exactly the same error message.
I'm not sure if there is a very basic C Coding error somewhere in my files.
You can see the whole code of the PWMServo example in my previous posts, but If it would help to have the whole package for testing, I gladly send it on request.

astagi, 2015-08-16:
yes please @balzreber send me your files I'll try to compile them directly and see what happens, you'll find my email on my profile (https://github.com/astagi)

balzreber, 2015-08-16:
@astagi thank you very much. I have sent you the whole shabang ;-)

astagi, 2015-08-20:
hey @balzreber I received your mail, thanks! I'll try to make it compile during the weekend!

astagi, 2015-08-23:
hey @balzreber here's the link of my Arduino folder: https://www.dropbox.com/s/vasuvh5obgqcxmx/Arduino_TLC5947.zip?dl=0

It now works! You made some mistakes using Nanpy (I'm sorry for that, it's not well documented :P), I'll list them below:

1) You forgot to use pointers, anytime you call "new" please create a new instance and pass it to the SlimArray

    if(strcmp(m->getName(), "new") == 0) {
        Adafruit_TLC5947* tlc = new Adafruit_TLC5947(1, 6, 7, 5);
        tlc->begin();
        v.insert(tlc);
        m->returns(v.getLastIndex());
    }

2) The ObjectManager needs to be init with the same type of the object used in the library, in this case *Adafruit_TLC5947*

   class TLC5947Class: public ObjectsManager<Adafruit_TLC5947>

3) You need to import Adafruit_TLC5947 in Nanpy.ino:

   #if USE_TLC5947
   #include <Adafruit_TLC5947.h>
   #endif

Hope it helps! If you want you may open a PR when it's ready and close this bug if this fixes your problem :)

closed 2015-08-29 by astagi

balzreber, 2015-08-30:
@astagi Brilliant!!! It works!!!
As I sed, id could very well be some basic stuf, I don't understand :-)
I made PR for both Nanpy and Nanpy-Firmware. Although I build the Nanpy-Firmware on the latest release on pypi as suggested. So I'm not sure if the Firmware works with the Git Release.
Thank you very much astagi for your kind help!