nanpy/nanpy #63: Add LCD createChar API

astagi, 2016-01-31:
An example

```
#include <LiquidCrystal.h>

LiquidCrystal lcd(12, 11, 5, 4, 3, 2);

byte smiley[8] = {
  B00000,
  B10001,
  B00000,
  B00000,
  B10001,
  B01110,
  B00000,
};

void setup() {
  lcd.createChar(0, smiley);
  lcd.begin(16, 2);  
  lcd.write(byte(0));
}

void loop() {}
```

astagi, 2016-02-14:
@joppi any update on this? need some help? :smiley_cat:

joppi, 2016-02-23:
@astagi I'm sorry, first I was busy with school stuff and then out of town. Should I push it straight to nanpy or go through pull request?

astagi, 2016-02-23:
oh no problem @joppi , just asking :smile: go for a PR, so I can review the code and make some tests easily.. Thank you!

astagi, 2016-02-24:
@joppi both PRs look awesome, thanks :smile: I'll test then tonight and merge!

closed 2016-02-27 by astagi