nanpy/nanpy #67: Getting an error when trying to instantiate a DallasTemperature object.

NeuralBlade, 2016-03-16:
Im running the code on an Arduino Mega and Python 2.7 running on OSX with Nanpy 0.9.6 installed via pip. Im no Python expert, so I have no clue what is going wrong...

Here is the code:
[nan.txt](https://github.com/nanpy/nanpy/files/176798/nan.txt)

Here is the output of the script.

> DEBUG:nanpy.serialmanager:opening port:/dev/cu.usbmodem1411 [115200 baud]
> DEBUG:nanpy.serialmanager:sending:'Info\x00'
> DEBUG:nanpy.serialmanager:sending:'0\x00'
> DEBUG:nanpy.serialmanager:sending:'0\x00'
> DEBUG:nanpy.serialmanager:sending:'count\x00'
> DEBUG:nanpy.serialmanager:received:'9\r\n'
> DEBUG:nanpy.serialmanager:sending:'Info\x00'
> DEBUG:nanpy.serialmanager:sending:'0\x00'
> DEBUG:nanpy.serialmanager:sending:'1\x00'
> DEBUG:nanpy.serialmanager:sending:'name\x00'
> DEBUG:nanpy.serialmanager:sending:'0\x00'
> DEBUG:nanpy.serialmanager:received:'A\r\n'
> DEBUG:nanpy.serialmanager:sending:'Info\x00'
> DEBUG:nanpy.serialmanager:sending:'0\x00'
> DEBUG:nanpy.serialmanager:sending:'1\x00'
> DEBUG:nanpy.serialmanager:sending:'name\x00'
> DEBUG:nanpy.serialmanager:sending:'1\x00'
> DEBUG:nanpy.serialmanager:received:'EEPROM\r\n'
> DEBUG:nanpy.serialmanager:sending:'Info\x00'
> DEBUG:nanpy.serialmanager:sending:'0\x00'
> DEBUG:nanpy.serialmanager:sending:'1\x00'
> DEBUG:nanpy.serialmanager:sending:'name\x00'
> DEBUG:nanpy.serialmanager:sending:'2\x00'
> DEBUG:nanpy.serialmanager:received:'RAM\r\n'
> DEBUG:nanpy.serialmanager:sending:'Info\x00'
> DEBUG:nanpy.serialmanager:sending:'0\x00'
> DEBUG:nanpy.serialmanager:sending:'1\x00'
> DEBUG:nanpy.serialmanager:sending:'name\x00'
> DEBUG:nanpy.serialmanager:sending:'3\x00'
> DEBUG:nanpy.serialmanager:received:'Lcd\r\n'
> DEBUG:nanpy.serialmanager:sending:'Info\x00'
> DEBUG:nanpy.serialmanager:sending:'0\x00'
> DEBUG:nanpy.serialmanager:sending:'1\x00'
> DEBUG:nanpy.serialmanager:sending:'name\x00'
> DEBUG:nanpy.serialmanager:sending:'4\x00'
> DEBUG:nanpy.serialmanager:received:'OneWire\r\n'
> DEBUG:nanpy.serialmanager:sending:'Info\x00'
> DEBUG:nanpy.serialmanager:sending:'0\x00'
> DEBUG:nanpy.serialmanager:sending:'1\x00'
> DEBUG:nanpy.serialmanager:sending:'name\x00'
> DEBUG:nanpy.serialmanager:sending:'5\x00'
> DEBUG:nanpy.serialmanager:received:'DallasTemperature\r\n'
> DEBUG:nanpy.serialmanager:sending:'Info\x00'
> DEBUG:nanpy.serialmanager:sending:'0\x00'
> DEBUG:nanpy.serialmanager:sending:'1\x00'
> DEBUG:nanpy.serialmanager:sending:'name\x00'
> DEBUG:nanpy.serialmanager:sending:'6\x00'
> DEBUG:nanpy.serialmanager:received:'D\r\n'
> DEBUG:nanpy.serialmanager:sending:'Info\x00'
> DEBUG:nanpy.serialmanager:sending:'0\x00'
> DEBUG:nanpy.serialmanager:sending:'1\x00'
> DEBUG:nanpy.serialmanager:sending:'name\x00'
> DEBUG:nanpy.serialmanager:sending:'7\x00'
> DEBUG:nanpy.serialmanager:received:'Core\r\n'
> DEBUG:nanpy.serialmanager:sending:'Info\x00'
> DEBUG:nanpy.serialmanager:sending:'0\x00'
> DEBUG:nanpy.serialmanager:sending:'1\x00'
> DEBUG:nanpy.serialmanager:sending:'name\x00'
> DEBUG:nanpy.serialmanager:sending:'8\x00'
> DEBUG:nanpy.serialmanager:received:'Info\r\n'
> Traceback (most recent call last):
>   File "nan.py", line 11, in <module>
>     sensors = DallasTemperature(2,a)
>   File "/usr/local/lib/python2.7/site-packages/nanpy/dallastemperature.py", line 9, in __init__
>     self.id = self.call('new', pin)
>   File "/usr/local/lib/python2.7/site-packages/nanpy/arduinoboard.py", line 150, in call
>     return _call(self.namespace, self.id, args, self.connection)
>   File "/usr/local/lib/python2.7/site-packages/nanpy/arduinoboard.py", line 28, in _call
>     _write(namespace, connection)
>   File "/usr/local/lib/python2.7/site-packages/nanpy/arduinoboard.py", line 9, in _write
>     connection.write('%s\0' % str(data))
> AttributeError: 'ArduinoApi' object has no attribute 'write'

dankozlowski, 2016-06-17:
Looks like a typo somwehere: `AttributeError: 'ArduinoApi' object has no attribute 'write'`.  Where are you calling #write?  Are you calling it on the correct instance of an object?

ponty, 2017-04-14:
DallasTemperature was called with wrong parameters.

closed 2017-04-14 by ponty