nanpy/nanpy #70: How can I detach a servo?

dankozlowski, 2016-06-17:
I have a couple of servos that need to remain open for an hour or so, and I'd like to disconnect them from their pins while they stay open.  Is there a way to do this with Nanpy?  I see a method to check Servo#attached and that #attach and #detach are implemented in the Firmware, but I can't seem to get attach or detach to work in Python.

omn0mn0m, 2017-03-22:
I know this is an old issue, but write this into servo.py:

```python
@arduinoobjectmethod
    def detach(self):
pass
```

Not sure why it is not in already.

dankozlowski, 2017-03-22:
Oh my God.  You wanna open a PR or shall I? :P

Thanks!

omn0mn0m, 2017-03-22:
Wait, did no one figure out this was missing...

ponty, 2017-04-14:
Merged.

closed 2017-04-14 by ponty