nanpy/nanpy #74: Wire Isssue

cheragnb, 2016-08-03:
Hi, 
Iv got Nanpy to do simple stuff such as moving motors and lighting LEDs (brilliant stuff btw ! ). Im trying to expand the IO ports of my Micro using a PCF8574 and i keep running into this error when i try the basic Demo program

>PCF8574 demo
address: 0x27
Traceback (most recent call last):
  File "PCF8574_demo.py", line 50, in <module>
    main()
  File "PCF8574_demo.py", line 15, in main
    pcf = PCF8574(a.wire, I2C_ADDRESS)
  File "/home/pi/Desktop/teststuff/nanpy/memo.py", line 20, in __call__
    value = self.func(*args)
  File "/home/pi/Desktop/teststuff/nanpy/arduinotree.py", line 141, in wire
    return Wire(self.connection)
  File "/home/pi/Desktop/teststuff/nanpy/classinfo.py", line 26, in getinstance
    (cls, cls_name))
nanpy.classinfo.FirmwareMissingFeatureError: <class 'nanpy.wire.Wire'> ['Wire'] is missing from firmware!

Any idea on what may be the problem ?
Thanks so much !

adamvisser, 2016-08-03:
> nanpy.classinfo.FirmwareMissingFeatureError: ['Wire'] is missing from

I am just getting into Nanpy myself, have you ensured that you properly
enabled the header files for 'Wire' before uploading the firmware? It seems
to be saying that it doesnt have the class file needed so it can run the
command you wanted?

I am probably way off bass here, but hopefully i have done something to
progess this issue lol

cheragnb, 2016-08-04:
Thanks @adam ! Yeah that did the trick !

closed 2016-08-04 by astagi