nanpy/nanpy #75: Interrupts

cheragnb, 2016-08-09:
Is there any library to enable a timer interrupt from either the Arduino or the raspberrypi ?

ponty, 2016-08-10:
Please write some details. What is it used for?

cheragnb, 2016-08-10:
A nanpy or similar port for a library such as timerOne. Where i can trigger an interrupt every 100 milliseconds. 
For me particularly im controlling a robot from a raspberry pi via an arduino. I need to update certain sensor values almost continuously (every 200 ms)

cheragnb, 2016-08-13:
I found a way to use interrupts, rather than interrupting the arduino, interrupt PYthon to updat eyour wrapper. 

def interrupt():
    threading.Timer(time_between_calls, interrupt).start()

make sure to include threading,time

ponty, 2016-08-16:
https://apscheduler.readthedocs.io/en/latest/

"Advanced Python Scheduler (APScheduler) is a Python library that lets you schedule your Python code to be executed later, either just once or periodically."

ponty, 2017-04-14:
Timer interrupt is not possible in nanpy firmware but it is possible in python.

closed 2017-04-14 by ponty