nanpy/nanpy #80: ServoShield

santhoshsandy1234, 2016-10-18:
I am a beginner for using nanpy. I need to communicate my Pi3 with Arduino UNO for controlling servos attached to my Arduino using ServoShield. I didnot find any nanpy library for my ServoShield and hopefully spent lot of time to extend library on my own but failed. I am running out of time to complete my project. I will be thankful if someone could extend the ServoShield library to nanpy  and it will definitely help everyone.
 The arduino library for my ServoShield is,
 https://github.com/santhoshsandy1234/ServoShield 

Please find solution for my library as soon as possible, It would help me a lot.
ThankYou.

ponty, 2016-10-21:
Why don't you use the built-in Servo library?
What is the source of your library?

santhoshsandy1234, 2016-10-22:
The ServoShield is has different library  called ServoShield.cpp I couldn't use Servo libray.
[ServoShield.zip](https://github.com/nanpy/nanpy/files/545737/ServoShield.zip)