nanpy/nanpy #81: Reading DHT11 Sensor Serial timeout

IvoVargas, 2016-11-09:
Hi,

Im using the example in https://github.com/nanpy/nanpy/blob/master/nanpy/examples/dht.py
I have a DHT11 sensor connected do digital pin 10.

And the result is this error:

Traceback (most recent call last):
  File "DHT.py", line 10, in <module>
    DHT(6, DHT.DHT11),
  File "/usr/local/lib/python2.7/dist-packages/nanpy/dht.py", line 13, in __init__
    self.id = self.call('new', pin, _type, count)
  File "/usr/local/lib/python2.7/dist-packages/nanpy/arduinoboard.py", line 150, in call
    return _call(self.namespace, self.id, args, self.connection)
  File "/usr/local/lib/python2.7/dist-packages/nanpy/arduinoboard.py", line 47, in _call
    ret = return_value(connection)
  File "/usr/local/lib/python2.7/dist-packages/nanpy/arduinoboard.py", line 18, in return_value
    return serial_manager.readline().replace('\r\n', '')
  File "/usr/local/lib/python2.7/dist-packages/nanpy/serialmanager.py", line 101, in readline
    raise SerialManagerError('Serial timeout!')
nanpy.serialmanager.SerialManagerError: Serial timeout!

I have tryed only "DHT(10, DHT.DHT11)", and have the same result.

Thanks

ponty, 2016-11-15:
Please turn on logging for more info:
```python
logging.basicConfig(level=logging.DEBUG)
```

kcicc77, 2016-11-26:
Did you figure it out? I plan on trying this example tomorrow.

pranjalnadhani, 2016-11-27:
I am facing the same error. I turned on logging and here is the dump:

> DEBUG:nanpy.serialmanager:opening port:/dev/ttyACM0 [115200 baud]
DEBUG:nanpy.serialmanager:sending:'DHT\x00'
DEBUG:nanpy.serialmanager:sending:'0\x00'
DEBUG:nanpy.serialmanager:sending:'3\x00'
DEBUG:nanpy.serialmanager:sending:'new\x00'
DEBUG:nanpy.serialmanager:sending:'6\x00'
DEBUG:nanpy.serialmanager:sending:'22\x00'
DEBUG:nanpy.serialmanager:sending:'6\x00'
DEBUG:nanpy.serialmanager:received:''
Traceback (most recent call last):
  File "dht_nanpy.py", line 13, in <module>
    DHT(6, DHT.DHT22),
  File "/usr/local/lib/python2.7/dist-packages/nanpy/dht.py", line 13, in __init__
    self.id = self.call('new', pin, _type, count)
  File "/usr/local/lib/python2.7/dist-packages/nanpy/arduinoboard.py", line 150, in call
    return _call(self.namespace, self.id, args, self.connection)
  File "/usr/local/lib/python2.7/dist-packages/nanpy/arduinoboard.py", line 47, in _call
    ret = return_value(connection)
  File "/usr/local/lib/python2.7/dist-packages/nanpy/arduinoboard.py", line 18, in return_value
    return serial_manager.readline().replace('\r\n', '')
  File "/usr/local/lib/python2.7/dist-packages/nanpy/serialmanager.py", line 101, in readline
    raise SerialManagerError('Serial timeout!')
nanpy.serialmanager.SerialManagerError: Serial timeout!

I hope that helps in figuring out the problem

ponty, 2016-11-28:
The Nanpy firmware doesn't respond.
It can be anything: hardware problem, bad cfg.h, wrong port,..

I created a 'bug reports' section in README, it may help you.

rndD, 2017-02-27:
Same with my Arduino Nano. 
How I can debug it? How I can be sure that DHT is enabled in firmware?

pranjalnadhani, 2017-03-01:
From the docs, you have to do the following:
```shell
git clone https://github.com/nanpy/nanpy-firmware.git
cd nanpy-firmware
```
You would get a file named `sample_cfg.h`. In that file, you would see the following lines:
```c
// https://github.com/adafruit/DHT-sensor-library
#define USE_DHT 0
```
You have to change the value of `USE_DHT` to 1, and also make sure you have the library from the [mentioned repository](https://github.com/adafruit/DHT-sensor-library). Once both the things are taken care of,
```shell
./configure.sh
```
Then you upload the Nanpy firmware to your Arduino.

ponty, 2017-03-02:
My DHT22 is working with Nano. Please read the README.

cfg.h:
```c
...
#define USE_DHT 1
...
```

Install in Arduino Library Manger:
 - Adafruit DHT sensor library
 - Adafruit Unified Sensor Library

python code:
```python
from nanpy import DHT
# http://learn.adafruit.com/dht/connecting-to-a-dhtxx-sensor
# DHT sensor connected to digital pin 10
dht = DHT(10, DHT.DHT22)
print("Humidity is %.2f %%" % dht.readHumidity())
```

rndD, 2017-03-05:
Oh my god. Thank you very much! Installation of `Adafruit Unified Sensor Library` is solved the problem. 
Sorry for wasting ur time.

closed 2017-03-11 by ponty

needhelpi, 2018-02-15:
hi there. how did you get the dht 11 sensor working? can i get it step by step? T_T
can anyone help please???