nanpy/nanpy #82: Servo Issue - appears to lose track of position

jsankovitch, 2016-12-14:
I'm not having a lot of luck getting a servo to work.  I'm running the example code modified to output debug logging.  The initial execution seems to work fine.  Subsequent ones leave it looking "lost" - as if it no longer remembers which position is which.

`from nanpy import Servo
import time
import logging

logging.basicConfig(level=logging.DEBUG)

servo = Servo(7)
for move in [0, 90, 180, 90, 0]:
    servo.write(move)
    time.sleep(1)
`

Debug from execution 1:
`DEBUG:nanpy.serialmanager:opening port:/dev/ttyACM0 [115200 baud]
DEBUG:nanpy.serialmanager:sending:'Servo\x00'
DEBUG:nanpy.serialmanager:sending:'0\x00'
DEBUG:nanpy.serialmanager:sending:'1\x00'
DEBUG:nanpy.serialmanager:sending:'new\x00'
DEBUG:nanpy.serialmanager:sending:'7\x00'
DEBUG:nanpy.serialmanager:received:'0\r\n'
DEBUG:nanpy.serialmanager:sending:'Servo\x00'
DEBUG:nanpy.serialmanager:sending:'0\x00'
DEBUG:nanpy.serialmanager:sending:'1\x00'
DEBUG:nanpy.serialmanager:sending:'write\x00'
DEBUG:nanpy.serialmanager:sending:'0\x00'
DEBUG:nanpy.serialmanager:received:'0\r\n'
DEBUG:nanpy.serialmanager:sending:'Servo\x00'
DEBUG:nanpy.serialmanager:sending:'0\x00'
DEBUG:nanpy.serialmanager:sending:'1\x00'
DEBUG:nanpy.serialmanager:sending:'write\x00'
DEBUG:nanpy.serialmanager:sending:'90\x00'
DEBUG:nanpy.serialmanager:received:'0\r\n'
DEBUG:nanpy.serialmanager:sending:'Servo\x00'
DEBUG:nanpy.serialmanager:sending:'0\x00'
DEBUG:nanpy.serialmanager:sending:'1\x00'
DEBUG:nanpy.serialmanager:sending:'write\x00'
DEBUG:nanpy.serialmanager:sending:'180\x00'
DEBUG:nanpy.serialmanager:received:'0\r\n'
DEBUG:nanpy.serialmanager:sending:'Servo\x00'
DEBUG:nanpy.serialmanager:sending:'0\x00'
DEBUG:nanpy.serialmanager:sending:'1\x00'
DEBUG:nanpy.serialmanager:sending:'write\x00'
DEBUG:nanpy.serialmanager:sending:'90\x00'
DEBUG:nanpy.serialmanager:received:'0\r\n'
DEBUG:nanpy.serialmanager:sending:'Servo\x00'
DEBUG:nanpy.serialmanager:sending:'0\x00'
DEBUG:nanpy.serialmanager:sending:'1\x00'
DEBUG:nanpy.serialmanager:sending:'write\x00'
DEBUG:nanpy.serialmanager:sending:'0\x00'
DEBUG:nanpy.serialmanager:received:'0\r\n'
`

Debug from execution 2:
`DEBUG:nanpy.serialmanager:opening port:/dev/ttyACM0 [115200 baud]
DEBUG:nanpy.serialmanager:sending:'Servo\x00'
DEBUG:nanpy.serialmanager:sending:'0\x00'
DEBUG:nanpy.serialmanager:sending:'1\x00'
DEBUG:nanpy.serialmanager:sending:'new\x00'
DEBUG:nanpy.serialmanager:sending:'7\x00'
DEBUG:nanpy.serialmanager:received:'1\r\n'
DEBUG:nanpy.serialmanager:sending:'Servo\x00'
DEBUG:nanpy.serialmanager:sending:'1\x00'
DEBUG:nanpy.serialmanager:sending:'1\x00'
DEBUG:nanpy.serialmanager:sending:'write\x00'
DEBUG:nanpy.serialmanager:sending:'0\x00'
DEBUG:nanpy.serialmanager:received:'0\r\n'
DEBUG:nanpy.serialmanager:sending:'Servo\x00'
DEBUG:nanpy.serialmanager:sending:'1\x00'
DEBUG:nanpy.serialmanager:sending:'1\x00'
DEBUG:nanpy.serialmanager:sending:'write\x00'
DEBUG:nanpy.serialmanager:sending:'90\x00'
DEBUG:nanpy.serialmanager:received:'0\r\n'
DEBUG:nanpy.serialmanager:sending:'Servo\x00'
DEBUG:nanpy.serialmanager:sending:'1\x00'
DEBUG:nanpy.serialmanager:sending:'1\x00'
DEBUG:nanpy.serialmanager:sending:'write\x00'
DEBUG:nanpy.serialmanager:sending:'180\x00'
DEBUG:nanpy.serialmanager:received:'0\r\n'
DEBUG:nanpy.serialmanager:sending:'Servo\x00'
DEBUG:nanpy.serialmanager:sending:'1\x00'
DEBUG:nanpy.serialmanager:sending:'1\x00'
DEBUG:nanpy.serialmanager:sending:'write\x00'
DEBUG:nanpy.serialmanager:sending:'90\x00'
DEBUG:nanpy.serialmanager:received:'0\r\n'
DEBUG:nanpy.serialmanager:sending:'Servo\x00'
DEBUG:nanpy.serialmanager:sending:'1\x00'
DEBUG:nanpy.serialmanager:sending:'1\x00'
DEBUG:nanpy.serialmanager:sending:'write\x00'
DEBUG:nanpy.serialmanager:sending:'0\x00'
DEBUG:nanpy.serialmanager:received:'0\r\n'
`

The servo is the only thing connected to my Arduino Uno.  I have also tried two different servos.  Any help is greatly appreciated.

dankozlowski, 2017-01-19:
Are you using a digital or analog servo?  I'm having a ton of trouble with digitals.

RyTheHackerGuy, 2017-05-25:
How did you even get the servos to do:
`servo = Servo(7)
servo.write(move)`
My code keeps throwing errors when I try to run either of these lines

dankozlowski, 2017-05-25:
@RyTheHackerGuy Going to need to see a lot more code than that to help.