nanpy/nanpy #83: Serial Timeout with Lcd

sikfrid, 2016-12-14:
Hi,
I'm trying to control an Arduino Uno with raspbian and python 3 but i have a problem with the lcd module, 
these lines:
`from nanpy import Lcd

lcd=Lcd( [7, 8, 9, 10, 11, 12], [16 , 2] )`

send back that error

Traceback (most recent call last):                                              
  File "<stdin>", line 1, in <module>                                           
  File "nanpy/lcd.py", line 12, in __init__                                     
      self.id = self.call('new', pins, begin)                                     
  File "nanpy/arduinoboard.py", line 150, in call                               
      return _call(self.namespace, self.id, args, self.connection)                
  File "nanpy/arduinoboard.py", line 47, in _call                               
      ret = return_value(connection)                                              
  File "nanpy/arduinoboard.py", line 18, in return_value                        
      return serial_manager.readline().replace('\r\n', '')                        
  File "nanpy/serialmanager.py", line 101, in readline                          
      raise SerialManagerError('Serial timeout!')                                 
nanpy.serialmanager.SerialManagerError: Serial timeout! 

However i can control pins using SerialManager() and ArduinoApi().

Thanks

jsankovitch, 2016-12-14:
Did you enable USE_LiquidCrystal_I2C in the cfg.h file before flashing the firmware to your arduino?

sikfrid, 2016-12-14:
Yes i did, but i just found out that some of the connections on the lcd board are bad, i'll do some soldering and see if it works better.

sikfrid, 2016-12-16:
That didn't change anything, i reflashed my arduino making sure the USE_LiquidCrystal_I2C was enabled, but i still have the same error.

ponty, 2016-12-16:
Please read the "Bug reports" section in README.

pietrondo, 2017-03-18:
same problem..

pietrondo, 2017-03-18:
code: 

`import string
import time
from nanpy import *
from nanpy import (OneWire, Lcd)
import logging
logging.basicConfig(level=logging.DEBUG)
# set up Hobbytronics Arduino LCD shield
lcd = Lcd([8, 9, 4, 5, 6, 7], [16, 2])

def scrollText( scrollBlurb ):
   if len(scrollBlurb) > 16:
     padding = " " * 16
     oldText = scrollBlurb
     scrollBlurb = padding + scrollBlurb + " "
     for i in range (0, len(scrollBlurb)):
      lcd.printString(scrollBlurb[i:(i+16)], 0, 1)
      time.sleep(0.35)
     lcd.printString(oldText[:16], 0, 1)
   else:
     lcd.printString(scrollBlurb, 0, 1)

scrollText("some very long text that won't fit on the screen")

`

log:

`DEBUG:nanpy.serialmanager:opening port:/dev/ttyUSB2 [115200 baud]
DEBUG:nanpy.serialmanager:sending:'Lcd\x00'
DEBUG:nanpy.serialmanager:sending:'0\x00'
DEBUG:nanpy.serialmanager:sending:'8\x00'
DEBUG:nanpy.serialmanager:sending:'new\x00'
DEBUG:nanpy.serialmanager:sending:'8\x00'
DEBUG:nanpy.serialmanager:sending:'9\x00'
DEBUG:nanpy.serialmanager:sending:'4\x00'
DEBUG:nanpy.serialmanager:sending:'5\x00'
DEBUG:nanpy.serialmanager:sending:'6\x00'
DEBUG:nanpy.serialmanager:sending:'7\x00'
DEBUG:nanpy.serialmanager:sending:'16\x00'
DEBUG:nanpy.serialmanager:sending:'2\x00'
DEBUG:nanpy.serialmanager:received:b''
Traceback (most recent call last):
  File "p.py", line 8, in <module>
    lcd = Lcd([8, 9, 4, 5, 6, 7], [16, 2])
  File "/usr/lib/python3.6/site-packages/nanpy/lcd.py", line 12, in __init__
    self.id = self.call('new', pins, begin)
  File "/usr/lib/python3.6/site-packages/nanpy/arduinoboard.py", line 150, in call
    return _call(self.namespace, self.id, args, self.connection)
  File "/usr/lib/python3.6/site-packages/nanpy/arduinoboard.py", line 47, in _call
    ret = return_value(connection)
  File "/usr/lib/python3.6/site-packages/nanpy/arduinoboard.py", line 18, in return_value
    return serial_manager.readline().replace('\r\n', '')
  File "/usr/lib/python3.6/site-packages/nanpy/serialmanager.py", line 101, in readline
    raise SerialManagerError('Serial timeout!')
nanpy.serialmanager.SerialManagerError: Serial timeout`

ponty, 2017-03-28:
You can embed Python code this way:

    ```python
    print 'hello'
    ```

ponty, 2017-04-01:
Please read the "Bug reports" section in README.

1. try the blink.py example
2. try to test the function without Nanpy, using only Arduino. If the problem remains then it is not a Nanpy bug.
3. enable logging: import logging;logging.basicConfig(level=logging.DEBUG)
4. attach the log messages
5. attach your program. The program should be as small as possible which demonstrates the bug.
6. check cfg.h. All needed functions are enabled?
7. attach your cfg.h
 

Missing in your reports:
- What is your hardware? HD44780 or I2C  interface?
- blink.py is working? -> hardware problems?
- does LCD work with Arduino example?
- cfg.h: USE_LiquidCrystal is set?