nanpy/nanpy #84: DS2482 via Nanpy I2C

cwynd, 2016-12-29:
Hi All, I am working on a distributed temperature and other data monitoring system, eventually to be used for control as well.

I'm using python (2.7) + Nanpy on a regular debian (jessie) PC (which will get me to a "high-level" back end for web, reporting and other stuff), which is talking to nanpy firmware on an arduino UNO in the same enclosure. This part all seems to be working well - I can 'python i2c_scan.py' and see the expected devices on the arduino I2C bus.

Next step is talking to a DS2482-800 I2C slave to 1-wire master. This also /had/ been going pretty well - I can talk to the DS2482 to the extent that I can send control codes etc with no errors, and if I intentionally send an invalid code, I get an error (NACK) from the I2C bus as expected.

But I cannot read logically valid data back from the DS2482 corresponding to the datasheet - whatever I send I always get the same reads back from the 4 internal registers. (But again no error on I2C bus).

I know it sounds like the DS2482 is fried or incorrectly configured, except I have successfully used this arduino library on the same unchanged hardware to interact with it:
https://github.com/paeaetech/paeae/tree/master/Libraries/ds2482
and find devices on the 1-wire side of the DS2482, so I really believe the hardware is good.

I have tried slowing the nanpy Serial down to 9600 with identical results (no errors), and I am beginning to think I am misunderstanding the data returned by I2C_Master.request, or how to use it. Especially as python is not my 'first language'.

The DS2482 requires setting a register pointer, and then issuing a read  on the I2C bus, and the only other thing I am wondering about is whether there is anything in the way the nanpy firmware sends and then reads that might disrupt that sequence - I notice that stop=False is not exposed in I2C_Master for example.

Would really appreciate any tips on where to look next to debug this setup. I can post my code if it's helpful obviously.

Thanks

cwynd, 2017-01-11:
Anyone out there??

cwynd, 2017-03-16:
In case anyone else ends up here, I decided that duplicating the DS2482 lib through nanpy and its I2C wrapper was not a smart way to go, and so I wrote a new nanpy + nanpy-firmware extension to wrap the existing arduino C library: https://github.com/paeaetech/paeae/tree/master/Libraries/ds2482

This is working really well, and I am datalogging from 1-wire devices connected to a DS2482 from python on desktop PC via arduino.

At some point I will commit and issue a pull request. If any specific interest pm me, and I will try to get to it sooner.

leofabridev, 2018-08-27:
Hi! I know that this topis is quite old but, are you still using Nanpy to read devices connected via I2C? I'm facing a lot of issues doing the same

cwynd, 2018-08-27:
Hi! Yes it is working well. We have a set up where a laptop (running our extended nanpy python code) scans via USB to the arduino and then I2C to the DS2482. We scan several 1-wire buses on the "external" side of the DS2482 every minute via a cron job, and write the results to a database, which can then be queried asynchronously by "higher level" software (we use cacti to make graphs). This has all been working with almost zero attention for the last year!