nanpy/nanpy #86: Controlling WS2812(B)/NeoPixel from nanpy?

ckildau, 2017-03-18:
Hi everyone,

I have an arduino nano hooked up to my notebook and wanted to start controlling an NeoPixel/LED  strip using nanpy. Any hints on where to start? I have not yet found a python library that is not depending on raspberry for generating the PWM signal.

omn0mn0m, 2017-03-24:
Hey, you should be able to use Nanpy to send functions to the Arduino from your laptop. It is not dependent on the Pi.

For PWM you can use analogWrite() from Nanpy.

ponty, 2017-04-01:
No support currently for WS2812, but existing WS2812 Arduino libraries can be added to Nanpy.
PWM is supported, check examples/pwm_demo.py

jc78, 2017-05-19:
I'm trying to do something very similar, but with Adafruit's DotStar strips. Unfortunately, I'm new to this stuff and only know a little Python. I don't know C\C++.

How would I use Adafruit's DotStar library along side nanpy? Ponty, are you referring to importing the C coded DotStar library into the nanpy firmware code? If so, what I don't understand is how to create the hooks from a serial message sent to nanpy and then access the DotStar functionality.

Or, do I import the Python DotStar library into my python code and somehow nanpy knows how to use it? (So I'm not touching any C code or changing the nanpy firmware.)

It would be great if somebody can point to an example of adding external libraries to Nanpy, extending it's capabilities. Sorry if this is super basic, like I said, I'm not familiar with C\C++ so I'm having a hard time trying to figure this out. In the Nanpy firmware code that I upload to my Arduino, the loop code is very simple:
```
void loop() {
    if(ComChannel::available()) {
        m = new MethodDescriptor();
        Register::elaborate(m);
    }
}
```
I'm assuming that I'll need to somehow modify this to know how to handle the new DotStar commands passed from Python?

jc78, 2017-05-19:
Actually, never mind, I just noticed the link in omn0mn0m's reply, to the documentation I'm looking for. I'll start there as it looks like it will answer all my questions.

Thanks!

SkyKnightOwl, 2017-08-08:
Can I check in and ask if did it work ? I'm working on a similar project and I'm having the same problems as you did.

Ekram-B, 2018-02-01:
Hey, @SkyKnightOwl, @jc78, were you successful? I'm also working on something similar

SkyKnightOwl, 2018-02-04:
@Ekram-B I settled on a different solution. Instead of using nanpy, what I did is set the arduino to receive data from a laptop using a pyserial. Then I put all the commands and the neopixel library to the arduino, created a code in a way where if I send a data (example "a") the arduino will respond based on the function that will be triggered from when it reads "a".

Ekram-B, 2018-02-04:
I actually decided to try a different solution as well. Let me provide a bit of context to what I'm doing. I'm trying to build a UMV (Unmanned Marine Vehicle) for my capstone. It's grown to become quite the system as we've had to buy many of the components to fit into our 1 year (to be more accurate 8 month) time constraint. One of the subsystems in our design involves use of a Raspberry Pi 3, and a Arduino Due (32 bit architecture). The arduino is meant to manage our sensors - (temperature sensor, pH sensor, and echo sounder, water collector) in accordance to a python script running on the Raspberry Pi. We've been able to use nanpy for our temperature sensor (DS18B20), and it's been really helpful for the pH sensor, and the water collector as well. For our echo sounder (ATU120), we were hoping to use an existing library written by Maarten Lamers in 2008 in C++. It was taking a bit of time extending Nanpy to do so, so we ended up writing up a bit of code where we connected our RPi to the arduino using the Serial library. Our issues have changed to the data that's supposedly being sent to our echo sounder not being sent back to the RPi. I'm exploring all sorts of things that could have caused this problem (maybe we aren't connecting our device properly but also maybe the code we've flashed to our Arduino device is written correctly). I admittedly am not an expert when it comes to the Arduino library. I was wondering if I could see a sample of the code that you described above? Maybe I could learn from it, and perhaps resolve our ongoing issue with our echo sounder.

SkyKnightOwl, 2018-02-05:
The code I made is not really that complex that may help you with your code but before I proceed, maybe a few tips as that may help you like you have to consider the time it takes before the serial connection is established before you can send and receive data from the rpi to arduino, the amount of data you can send depends on the baud rate you have set and it may take some time before you receive any data so maybe a pause in the program can help?, and lastly maybe you can try using both serial and nanpy like create txts and a python that can read the txts so you can transfer data if ever you need data from nanpy and data from a serial connection.