nanpy/nanpy #91: DHT, Servo, LiquidCrystal give a Serial Timeout error

pranjalnadhani, 2017-05-10:
Whenever I try to connect any of the DHT, Servo or LiquidCrystal hardware, I get a Serial Timeout error. Here is a sample program for DHT:

```shell
Python 2.7.12 (default, Nov 19 2016, 06:48:10) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from nanpy import DHT
>>> import logging
>>> logging.basicConfig(level=logging.DEBUG)
>>> dht = DHT(8, DHT.DHT11)
DEBUG:nanpy.serialmanager:opening port:/dev/ttyACM0 [115200 baud]
DEBUG:nanpy.serialmanager:sending:'DHT\x00'
DEBUG:nanpy.serialmanager:sending:'0\x00'
DEBUG:nanpy.serialmanager:sending:'3\x00'
DEBUG:nanpy.serialmanager:sending:'new\x00'
DEBUG:nanpy.serialmanager:sending:'8\x00'
DEBUG:nanpy.serialmanager:sending:'11\x00'
DEBUG:nanpy.serialmanager:sending:'6\x00'
DEBUG:nanpy.serialmanager:received:''
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pranjal/.local/lib/python2.7/site-packages/nanpy/dht.py", line 13, in __init__
    self.id = self.call('new', pin, _type, count)
  File "/home/pranjal/.local/lib/python2.7/site-packages/nanpy/arduinoboard.py", line 150, in call
    return _call(self.namespace, self.id, args, self.connection)
  File "/home/pranjal/.local/lib/python2.7/site-packages/nanpy/arduinoboard.py", line 47, in _call
    ret = return_value(connection)
  File "/home/pranjal/.local/lib/python2.7/site-packages/nanpy/arduinoboard.py", line 18, in return_value
    return serial_manager.readline().replace('\r\n', '')
  File "/home/pranjal/.local/lib/python2.7/site-packages/nanpy/serialmanager.py", line 101, in readline
    raise SerialManagerError('Serial timeout!')
nanpy.serialmanager.SerialManagerError: Serial timeout!
>>> 
```
My `cfg.h` file is as follows:

```c
#pragma once

// only for ESP8266
#define USE_WIFI_CONNECTION                         0

// for WIFI connection
#define WIFI_SSID 		"***"
#define WIFI_PASSWORD 	"***"

// for serial connection
#define BAUDRATE 115200
#define COMM_SERIAL                                 Serial

// info about existence of other features
#define USE_Info                                    1

// definitions: MCU type, frequency, Arduino version,
// EEPROM size, RAM size, pin count, build time,..
#define USE_Define                                  1

// low level mapping of pins and ports
#define USE_ArduinoCore                             1

// read, write RAM
#define USE_RAM                                     1

// read, write EEPROM
#define USE_EEPROM                                  1

// read, write AVR registers
#define USE_Register                                0

// watchdog and reset
#define USE_Watchdog                                0

#define USE_Tone                                    0

#define USE_LiquidCrystal                           1

// I2C
#define USE_Wire                                    0

#define USE_Servo                                   1

#define USE_Stepper                                 0

// frequency counter,  USE_Tone should be off!
#define USE_Counter                                 0

// GW Robotics Classes
#define USE_Ultrasonic				                0
#define USE_ColorSensor                             0

///////////////////////////////////////////////////////////////////////
// external libraries should be installed for the following features:
///////////////////////////////////////////////////////////////////////

// install with Library Manager: "OneWire"
// https://github.com/PaulStoffregen/OneWire
#define USE_OneWire                                 0

// install with Library Manager: "DallasTemperature"
// https://github.com/milesburton/Arduino-Temperature-Control-Library
#define USE_DallasTemperature                       0

// install with Library Manager: "CapacitiveSensor"
// https://github.com/PaulStoffregen/CapacitiveSensor
#define USE_CapacitiveSensor                        0

// install with Library Manager: "DHT sensor library","Adafruit Unified Sensor"
// https://github.com/adafruit/DHT-sensor-library
#define USE_DHT                                     1

// https://bitbucket.org/fmalpartida/new-liquidcrystal
#define USE_LiquidCrystal_I2C                       0

// install with Library Manager: "Adafruit TLC5947"
// https://github.com/adafruit/Adafruit_TLC5947
#define USE_TLC5947                                 0

// install with Library Manager: "MCP41XXX"
//     latest version is bad (1.0.3), it should be installed manually
// https://github.com/ArduinoMax/MCP41xxx
#define USE_MCP41xxx                                0
```

In the case of DHT, I have the **DHT Sensor Library** and **Adafruit Unified Sensor Library** installed. They are working independently in the Arduino. Similar thing happens with Servo and LiquidCrystal. However, this piece of code works as expected:

```shell
Python 2.7.12 (default, Nov 19 2016, 06:48:10) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from nanpy import (ArduinoApi, SerialManager)
>>> con = SerialManager("/dev/ttyACM0")
>>> a = ArduinoApi(connection = con)
>>> a.pinMode(13, a.OUTPUT)
u'0'
>>> a.digitalWrite(13, a.HIGH)
u'0'
>>> a.digitalWrite(13, a.LOW)
u'0'
>>> 
```

ponty, 2017-05-10:
Your cfg.h is OK, nanpy works with it. 
I suspect that you have the wrong firmware without DHT support.
Run examples/dump.py and send the results.

pranjalnadhani, 2017-05-12:
I ran the examples/dump.py program and here are the results:

```shell
Traceback (most recent call last):
  File "dump.py", line 7, in <module>
    from nanpy.classinfo import FirmwareMissingFeatureError
ImportError: cannot import name FirmwareMissingFeatureError
```

ponty, 2017-05-14:
You are using 1 year old version of Nanpy.

pranjalnadhani, 2017-05-14:
I installed nanpy using the pip installer. The date of last updated shows `2016-03-04`. Do I need to manually build the library from the repo? Or is there a workaround using pip?

ponty, 2017-05-14:
The pip version is old. Use the github version.

pranjalnadhani, 2017-05-14:
Okay. Can you provide the instructions to install nanpy via github? I'm having trouble finding those. I'm only getting the guide to install the nanpy-firmware wherever I search.

pranjalnadhani, 2017-05-15:
I could finally install the latest version of nanpy from GitHub. The Servo is now working perfectly. But when I try to use the DHT, I get the following error:

```shell
Traceback (most recent call last):
  File "dht.py", line 9, in <module>
    dht = DHT(10, DHT.DHT22)
  File "build/bdist.linux-x86_64/egg/nanpy/dht.py", line 14, in __init__
    
  File "build/bdist.linux-x86_64/egg/nanpy/arduinoboard.py", line 150, in call
  File "build/bdist.linux-x86_64/egg/nanpy/arduinoboard.py", line 47, in _call
  File "build/bdist.linux-x86_64/egg/nanpy/arduinoboard.py", line 18, in return_value
  File "build/bdist.linux-x86_64/egg/nanpy/serialmanager.py", line 101, in readline
nanpy.serialmanager.SerialManagerError: Serial timeout!
```

I have attached a DHT22 in pin 10 of my Arduino Uno board. I tried running DHT test on the Arduino using the classic Arduino code, and it works perfectly.

ponty, 2017-05-17:
I suspect that you have the wrong firmware without DHT support.
Run examples/dump.py to check it.

ponty, 2017-05-25:
Please read ISSUE_TEMPLATE.md.

Jibachi, 2017-12-12:
I have the same problem with the sample dht.py with my DHT22 sensor and I followed the ISSUE_TEMPLATE.md until the fourth bullet point:

1. blink.py works
2. Arduino code works
3. cfg.h in nanpy-firmware in the library folder of Arduino IDE looks like this:
```
#pragma once

// only for ESP8266
#define USE_WIFI_CONNECTION                         0

// for WIFI connection
#define WIFI_SSID 		"***"
#define WIFI_PASSWORD 	"***"

// for serial connection
#define BAUDRATE 115200
#define COMM_SERIAL                                 Serial

// info about existence of other features
#define USE_Info                                    1

// definitions: MCU type, frequency, Arduino version,
// EEPROM size, RAM size, pin count, build time,..
#define USE_Define                                  1

// low level mapping of pins and ports
#define USE_ArduinoCore                             1

// read, write RAM
#define USE_RAM                                     1

// read, write EEPROM
#define USE_EEPROM                                  1

// read, write AVR registers
#define USE_Register                                0

// watchdog and reset
#define USE_Watchdog                                0

#define USE_Tone                                    0

#define USE_LiquidCrystal                           0

// I2C
#define USE_Wire                                    0

#define USE_Servo                                   1

#define USE_Stepper                                 0

// frequency counter,  USE_Tone should be off!
#define USE_Counter                                 0

// GW Robotics Classes
#define USE_Ultrasonic				                0
#define USE_ColorSensor                             0

///////////////////////////////////////////////////////////////////////
// external libraries should be installed for the following features:
///////////////////////////////////////////////////////////////////////

// install with Library Manager: "OneWire"
// https://github.com/PaulStoffregen/OneWire
#define USE_OneWire                                 0

// install with Library Manager: "DallasTemperature"
// https://github.com/milesburton/Arduino-Temperature-Control-Library
#define USE_DallasTemperature                       0

// install with Library Manager: "CapacitiveSensor"
// https://github.com/PaulStoffregen/CapacitiveSensor
#define USE_CapacitiveSensor                        0

// install with Library Manager: "DHT sensor library","Adafruit Unified Sensor"
// https://github.com/adafruit/DHT-sensor-library
#define USE_DHT                                     1

// https://bitbucket.org/fmalpartida/new-liquidcrystal
#define USE_LiquidCrystal_I2C                       0

// install with Library Manager: "Adafruit TLC5947"
// https://github.com/adafruit/Adafruit_TLC5947
#define USE_TLC5947                                 0

// install with Library Manager: "MCP41XXX"
// https://github.com/ArduinoMax/MCP41xxx
#define USE_MCP41xxx                                0

```

4. Running dump.py looks like this:
```
===============
['ArduinoApi', 'DefineArray', 'ClassInfoArray', 'RAM', 'Servo']

================================
defines:
================================
A0                   =                   14
ARDUINO              =                  105
E2END                =                 1023
E2PAGESIZE           =                    4
FLASHEND             =                32767
F_CPU                =             16000000
MCU                  =   __AVR_ATmega328P__
NUM_DIGITAL_PINS     =                   20
RAMEND               =                 2303
RAMSTART             =                  256
SPM_PAGESIZE         =                  128
XRAMEND              =                 2303
__AVR_LIBC_DATE_     =             20111229
__AVR_LIBC_VERSION__ =                10800
__DATE__             =          Dec 11 2017
__TIME__             =             18:13:58
__VERSION__          =                4.9.2

================================
pins:
================================
total_pin_count      =                   20
digital_names        = ['D0', 'D1', 'D2', 'D3', 'D4', 'D5', 'D6', 'D7', 'D8', 'D9', 'D10', 'D11', 'D12', 'D13']
analog_names         = ['A0', 'A1', 'A2', 'A3', 'A4', 'A5']
---------- pin_number=0 ---------------
name                 =                   D0
pin_number           =                    0
pin_number_analog    =                 None
is_digital           =                 True
is_analog            =                False
Traceback (most recent call last):
  File "dump.py", line 121, in <module>
    dumpall()
  File "dump.py", line 93, in dumpall
    'name pin_number pin_number_analog is_digital is_analog avr_pin mode digital_value analog_value programming_function'.split())
  File "dump.py", line 21, in dump
    if not inspect.ismethod(getattr(obj, attr)):
  File "/usr/local/lib/python3.5/dist-packages/nanpy-0.9.6-py3.5.egg/nanpy/arduinopin.py", line 115, in avr_pin
  File "/usr/local/lib/python3.5/dist-packages/nanpy-0.9.6-py3.5.egg/nanpy/arduinopin.py", line 96, in avr_port
AttributeError: 'NoneType' object has no attribute 'digitalPinToPort'

```
4. Running firmware_check.py looks like this
```
Firmware classes enabled in cfg.h:
  ArduinoApi
  DefineArray
  ClassInfoArray
  RAM
  Servo

Your firmware was built on:
  Dec 11 2017 18:13:58
```

I think the DHT support wasn't uploaded to the arduino, but I don't know why. DHT Sensor Library and Adafruit Unified Sensor Library was copied into the library folder of Arduino IDE.

GuyFawkes89, 2018-01-14:
I had the same issue (Serial Timeout) with the LiquidCrystal_I2C.
Try to edit also cfg_defaults.h and cfg_check.h enabling the libraries you need.
It worked for me!