nanpy/nanpy #92: Using Servos

RyTheHackerGuy, 2017-05-25:
I cant figure out the Servo usage, this is the shell error:

```
Traceback (most recent call last):
  File "/home/pi/Desktop/nanpy/RobotCode.py", line 28, in <module>
    servoA = Servo(2)
  File "/usr/local/lib/python3.4/dist-packages/nanpy-0.9.6-py3.4.egg/nanpy/servo.py", line 9, in __init__
    self.id = self.call('new', pin)
  File "/usr/local/lib/python3.4/dist-packages/nanpy-0.9.6-py3.4.egg/nanpy/arduinoboard.py", line 150, in call
    return _call(self.namespace, self.id, args, self.connection)
  File "/usr/local/lib/python3.4/dist-packages/nanpy-0.9.6-py3.4.egg/nanpy/arduinoboard.py", line 47, in _call
    ret = return_value(connection)
  File "/usr/local/lib/python3.4/dist-packages/nanpy-0.9.6-py3.4.egg/nanpy/arduinoboard.py", line 18, in return_value
    return serial_manager.readline().replace('\r\n', '')
  File "/usr/local/lib/python3.4/dist-packages/nanpy-0.9.6-py3.4.egg/nanpy/serialmanager.py", line 101, in readline
    raise SerialManagerError('Serial timeout!')
nanpy.serialmanager.SerialManagerError: Serial timeout!
```

here is my code:
...
from nanpy import (ArduinoApi, SerialManager)
import pygame
from nanpy import Servo
import time
pygame.init()

a=0
d=0

window = pygame.display.set_mode((800,600))

pygame.display.set_caption("Console")

try :
    connection = SerialManager()
    ard = ArduinoApi(connection = connection)
    run = True
except:
    print("Connection Failed!")

servoA = Servo(2)
servoD = Servo(4)
while run :
      for event in pygame.event.get():
                     keys = pygame.key.get_pressed()
           # if (event.type==pygame.KEYDOWN):

                     if keys[pygame.K_s] and keys[pygame.K_a]:
                            a=a-1
                            servoA.write(a)
                     elif keys[pygame.K_s] and keys[pygame.K_d]:
                            d=d-1
                            servoD.write(d)

                     elif keys[pygame.K_w]:
                            a=a+1
                            d=d+1
                            servoD.write(d)
                            servoA.write(a)

                     elif keys[pygame.K_s]:
                            a=a-1
                            d=d-1
                            servoD.write(d)
                            servoA.write(a)

                     elif keys[pygame.K_d]:

                            d=d+1
                            servoD.write(d)

                     elif keys[pygame.K_a]:
                            a=a+1
                            servoA.write(a)

                     elif keys[pygame.K_t]:
                            run=False

pygame.quit()

...

dankozlowski, 2017-05-25:
Serial timeout means you likely didn't bootstrap the Arduino successfully.

RyTheHackerGuy, 2017-05-25:
@dankozlowski What do I need to do to fix the Bootstrap?

ponty, 2017-05-25:
Please read ISSUE_TEMPLATE.md.
Please remove Pygame from your code and make it simple.

dankozlowski, 2017-05-25:
@RyTheHackerGuy I'd start by posting steps to reproduce.

dankozlowski, 2017-05-25:
How did you bootstrap you Arduino?  Can you paste the output from the programmer?

RyTheHackerGuy, 2017-05-25:
@ponty  pygame is for tracking the buttons being pressed, I found it to be easier than using tkinter

RyTheHackerGuy, 2017-05-25:
@dankozlowski I followed exactly as shown in this video: [](https://www.youtube.com/watch?v=QumIhvYtRKQ)

RyTheHackerGuy, 2017-05-25:
As I'm still very new to this, I kinda assumed the the Serialconnection part of the code made the bootstrap  (I have never worked with a Pi or Arduino before this project)

ponty, 2017-05-25:
README: "You can now edit Nanpy/cfg.h generated file to configure your Nanpy firmware, selecting the features you want to include and the baud rate."

Edit the cfg.h file and enable Servo.

RyTheHackerGuy, 2017-05-25:
@ponty I think you saved me bro

RyTheHackerGuy, 2017-05-25:
oh and just as a clarifying question, 
'servoOne =Servo(1)'
binds to the digital output right?