nanpy/nanpy #94: Max USB Speed between Raspi3 and Arduino Mega 2560

sTingAnet, 2017-06-03:
Hey,

at first, very nice work! I love it.
At secound everything works fine, but not fast enough :D ...

I think my setup is a hardcore project for nanpy.
I have 32 digital inpus, 24 digital outputs, 8 analog inputs and 2x i2c/wire connection to 2 additional arduinos with each 4 analog outputs (0-255). I have a 2 thread python code to handle nanpy in one thread and my application a the secound. The working speed for the nanpy thread is ~250ms. so each secound the nanpy connection between the Raspi3 and the arduino 2560 handels the I/Os four times. I have set the code to in minimum, i think ^^. Only if a (analog (i2C) or digital) output has to change i send the change, but every loop nanpy has to read the inputs form the arduino (digital and analog). I've done this with "for".

Code:
`
            for x in range(54,62):
                namespace['state_ap%d' % x] = a.analogRead(x)
            for x in range(14,20):
                namespace['state_ap%d' % x] = a.digitalRead(x)
            for x in range(23,55,2):
                namespace['state_ap%d' % x] = a.digitalRead(x)
            for x in range(2,12):
                namespace['state_ap%d' % x] = a.digitalRead(x)
`

I set the baudrate for the connection from default 115300 to 230400 (in napy firmeware and in my python nanpy thread by hand), connection works with 230400,  but the code-working-speed isn't geeting faster, so the USB connection ins't the bottleneck.

So any ideas were is the bottleneck?
Does the for-loop does a "connection" to the arduino, reads the value and "close" the "connection", so that takes a lot of time to get this one by one ?
Or is it a hardware bottleneck, maybe between the Atmel-USB-Programmer(MEGA16U2) and the Mega2560?
Or do i have to shrink the nanpy-firmware to get more speed?

If more code is requiert, i've >1300 lines to this point :D, ok only 80 lines are for the nanpy connection...
Thank you for an answer :)

sTingAnet, 2017-06-06:
Hey,

i've done some investigation.

my hole Python Script without Nanpy-Connection works with 0,5ms for one loop, yeah thats too fast :D ... 20 loops per Secound would be enough for my project -> 50ms.

One single digitalRead request from python to the Arduino takes 4ms.
The hole communication is linear, so every additional digitalRead request takes 4ms.
1 request = 4ms
2  requests = 8ms
5  requests = 20ms
10  requests = 40ms
20 requests = 80ms
30 requests = 120ms
100 requests = 400ms

That seems to me, every Operation is handled once, before the next one starts ?

The Serial/USB Speed itself again can't be the issue, increasing the baudrate doesn't take any effect.
The Speed of the arduino can't also be the Problem, I think?! 
16Mhz should be better than 4ms for one Arduino loop, even if I don't calculate/know the loop time for the Nanpy-Firmware-Code, the Arduino (Atmega) should be capable to handle this faster than 4ms. 16Mhz are 64.000 Ticks in 4ms. I can't imagine that loop on the Atmega takes that time. 
The Atmega 2560 can Print 3D Objects in realtime with Serial-Commucation to my PC, calculate remaining Time and show Temperatures on the LCD, toggle a PID for the Temperatures and so on and on ... so what is going on here?
I checked the Datasheet of the 16u2, full USB2.0 Capability.
I also tested a serial connection with 230400 and above that baudrate with a Test Programm and it sends at that speed, no Problem.

I do not understand it, sorry.
If nobody has an idea, i've to build the hole communication on my own to get more speed.

ponty, 2017-06-30:
You can check the communication if you turn on logging with timestamps.

```python
from nanpy import (SerialManager, ArduinoApi)
import logging

FORMAT = '%(asctime)-6s: %(name)s - %(levelname)s - %(message)s'
logging.basicConfig(level=logging.DEBUG, format=FORMAT)

connection = SerialManager()
a = ArduinoApi(connection=connection)

for x in range(0, 4):
    print a.analogRead(x)
```

I got similar results: 1 request = 4ms

```
2017-06-30 22:53:16,197: nanpy.serialmanager - DEBUG - sending:'0\x00'
2017-06-30 22:53:16,197: nanpy.serialmanager - DEBUG - sending:'1\x00'
2017-06-30 22:53:16,197: nanpy.serialmanager - DEBUG - sending:'a\x00'
2017-06-30 22:53:16,197: nanpy.serialmanager - DEBUG - sending:'1\x00'
2017-06-30 22:53:16,200: nanpy.serialmanager - DEBUG - received:'292\r\n'
292
2017-06-30 22:53:16,201: nanpy.serialmanager - DEBUG - sending:'A\x00'
2017-06-30 22:53:16,201: nanpy.serialmanager - DEBUG - sending:'0\x00'
2017-06-30 22:53:16,201: nanpy.serialmanager - DEBUG - sending:'1\x00'
2017-06-30 22:53:16,201: nanpy.serialmanager - DEBUG - sending:'a\x00'
2017-06-30 22:53:16,201: nanpy.serialmanager - DEBUG - sending:'2\x00'
2017-06-30 22:53:16,205: nanpy.serialmanager - DEBUG - received:'303\r\n'
303
```

The response is slower than I expect:
communication: 5*10/100k=0.5ms
ADC: 0.1ms

I suspect that the firmware code can be optimized for performance.

sTingAnet, 2017-07-04:
Hey, thank you for your response.
I've done the communication by my own, so i don't use nanpy for this project anymore.
But maybe some performance improvements are also good for other users.