nanpy/nanpy #95: Extending nanpy with Adafruit's DotStar Library

jc78, 2017-06-05:
I'm trying to extend nanpy by supporting Adafruit's DotStar LED strips and I'm running into a serial timeout when I'm trying to create an instance of the DotStar object I added.

I followed the instructions [found here](https://nanpy.github.io/#how-to-extend-nanpy), and was able to get my sketch to compile and upload to my Uno board without any issues. I have run the blink.py test afterwards to ensure the nanpy firmware is working as expected and it is. I also tried running Adafruit's strandtest.ino to ensure that all of the hardware is functioning as expected, and it is.

My problem happens when I try to create a DotStar object in my python code. I'm assuming that the Adafruit_DotStar class I made is failing in some way. I'm not proficient in C++ so the best I could do was follow the documents and hope for the best. :-) I ensured that the actual Adafruit_Dotstar library is in my Arduino Sketchbook\Libraries folder so it can be included properly.

### Your Python code

```
from nanpy import (ArduinoApi, SerialManager, dotstar)
import time
import logging
logging.basicConfig(level=logging.DEBUG)

connection = SerialManager()
arduino = ArduinoApi(connection=connection)

strip = dotstar.DotStar(3, 4, 5, connection=connection)

while True:
    for x in range(3):
        strip.setPixelColor(x, [0,0,0])
    time.sleep(0.5)
    
    for x in range(3):
        strip.setPixelColor(x, [255,0,0])
    time.sleep(0.5)
```

### Your log messages

```
Python 2.7.9 (default, Sep 17 2016, 20:26:04) 
[GCC 4.9.2] on linux2
Type "copyright", "credits" or "license()" for more information.
>>> ================================ RESTART ================================
>>> 
DEBUG:nanpy.serialmanager:opening port:/dev/ttyACM0 [115200 baud]
DEBUG:nanpy.serialmanager:sending:'Info\x00'
DEBUG:nanpy.serialmanager:sending:'0\x00'
DEBUG:nanpy.serialmanager:sending:'0\x00'
DEBUG:nanpy.serialmanager:sending:'count\x00'
DEBUG:nanpy.serialmanager:received:'8\r\n'
DEBUG:nanpy.serialmanager:sending:'Info\x00'
DEBUG:nanpy.serialmanager:sending:'0\x00'
DEBUG:nanpy.serialmanager:sending:'1\x00'
DEBUG:nanpy.serialmanager:sending:'name\x00'
DEBUG:nanpy.serialmanager:sending:'0\x00'
DEBUG:nanpy.serialmanager:received:'A\r\n'
DEBUG:nanpy.serialmanager:sending:'Info\x00'
DEBUG:nanpy.serialmanager:sending:'0\x00'
DEBUG:nanpy.serialmanager:sending:'1\x00'
DEBUG:nanpy.serialmanager:sending:'name\x00'
DEBUG:nanpy.serialmanager:sending:'1\x00'
DEBUG:nanpy.serialmanager:received:'RAM\r\n'
DEBUG:nanpy.serialmanager:sending:'Info\x00'
DEBUG:nanpy.serialmanager:sending:'0\x00'
DEBUG:nanpy.serialmanager:sending:'1\x00'
DEBUG:nanpy.serialmanager:sending:'name\x00'
DEBUG:nanpy.serialmanager:sending:'2\x00'
DEBUG:nanpy.serialmanager:received:'D\r\n'
DEBUG:nanpy.serialmanager:sending:'Info\x00'
DEBUG:nanpy.serialmanager:sending:'0\x00'
DEBUG:nanpy.serialmanager:sending:'1\x00'
DEBUG:nanpy.serialmanager:sending:'name\x00'
DEBUG:nanpy.serialmanager:sending:'3\x00'
DEBUG:nanpy.serialmanager:received:'Core\r\n'
DEBUG:nanpy.serialmanager:sending:'Info\x00'
DEBUG:nanpy.serialmanager:sending:'0\x00'
DEBUG:nanpy.serialmanager:sending:'1\x00'
[... 365 lines cut ...]
    REGISTER_CLASS_CONDITIONAL(WireClass, USE_Wire);
    
    REGISTER_CLASS_CONDITIONAL(TLC5947Class, USE_TLC5947);

    REGISTER_CLASS_CONDITIONAL(nanpy::EspClass, USE_ESP);

    REGISTER_CLASS_CONDITIONAL(MeasureClass, USE_Measure);
    REGISTER_CLASS_CONDITIONAL(DotStarClass, USE_DotStar);
    
    ComChannel::connect();
}

void loop() {
    if(ComChannel::available()) {
        m = new MethodDescriptor();
        Register::elaborate(m);
    }
}
```

### Your hardware

- Arduino board: `Uno`
- Additional hardware:  `Adafruit DotStar LED Strip (only testing with 3 pixels (LEDs)`
- `Verified the DotStar Strip is properly wired.`

jc78, 2017-06-07:
Beuller? Beuller?

Anyone here have any ideas or suggestions?

jc78, 2017-06-19:
Are there any nanpy devs in here that can help out? I've been trying in vein to make this work but with no success.

@astagi 
@ponty

ponty, 2017-06-30:
You use different identifier in Python and C++ code:
C++: Adafruit_DotStar
Python: DotStar

You can see it in the log.

Ekram-B, 2018-02-01:
Hi,

I was wondering if @jc78 has their issue resolved. I'm trying to extend nanpy to accomodate Marten Lamer's NMEA library to nanpy. I'm running into a similar issue where I experience a SerialManager timeout.