nanpy/nanpy #97: Running the combined arduino and pi python code.

bdavis721, 2017-07-01:
Great job, My thing is more of a question than an issue I just didnt know where to put it. Here are my questions:

When I run my python script, do I need a serial connection even if I dont need the serial monitor?
Do I need to run the arduino part of the code on the arduino ide and the pi code on the pi or can it all be run from the pi as one script?
Do I need to import Arduino or ArdunioApi from nanpy library?

Thanks

ponty, 2017-07-01:
Nanpy-firmware runs on Arduino.
Your Python script and Nanpy Python library run on Pi (or PC)
The 2 programs communicate over serial connection.
Check the README for installation and examples.

bdavis721, 2017-07-04:
I tried to run the code and got the error 

> Traceback (most recent call last):
  File "SmartController.py", line 13, in <module>
    a = ArduinoApi()
TypeError: getinstance() takes exactly 1 argument (0 given)

But in the README the first section doesn't give an argument.

ololobus, 2017-07-04:
@bdavis721 Please, refer to the next section https://github.com/nanpy/nanpy#serial-communication
It states 'Nanpy autodetects the serial port for you, anyway you can manually specify another serial port'

I guess that autodetect can fail in some cases (e.g. mine), so I simply specify serial port manually and it works well