nanpy/nanpy #109: SerialTimeout with I2C and Sunfounder LCD 2004a

Desperandum, 2019-03-25:
I'm attempting to use a raspberry pi to send information to an LCD screen connected to an arduino. Using Arduino IDE I can make text appear and use regular LiquidCrystal library functions, but when attempting to use Nanpy I get serial timeouts. Below is information regarding these errors. I can successfully run the i2c_scanner and blink examples from nanpy, but I cannot run firmware_check or lcd_hello examples. It might be from my pin setup, which I'm not sure what is or how to check (bought off Amazon, not ebay). Address checks out to be 0x27.

- [ X] Try `examples/blink.py`
- [ X] Try to test the function without Nanpy, using only Arduino code.
- [ X] Check cfg.h. All needed functions are enabled?
- [ Failed to check] Run `examples/firmware_check.py` to check cfg.h again. All needed functions are listed?

Using examples/lcd_hello.py with cols,rows = 20, 4

Output of lcd_hello.py:

```
Traceback (most recent call last):
  File "./lcd_hello.py", line 27, in <module>
    hello()
  File "./lcd_hello.py", line 17, in hello
    lcd = Lcd_I2C(pins, [cols, rows], connection=connection)
  File "build/bdist.linux-armv7l/egg/nanpy/lcd_i2c.py", line 14, in __init__
  File "build/bdist.linux-armv7l/egg/nanpy/arduinoboard.py", line 150, in call
  File "build/bdist.linux-armv7l/egg/nanpy/arduinoboard.py", line 47, in _call
  File "build/bdist.linux-armv7l/egg/nanpy/arduinoboard.py", line 16, in return_value
  File "build/bdist.linux-armv7l/egg/nanpy/serialmanager.py", line 101, in readline
nanpy.serialmanager.SerialManagerError: Serial timeout!
```

Output of firmware_check:

```
from: can't read /var/mail/nanpy.arduinotree
from: can't read /var/mail/nanpy.serialmanager
./firmware_check.py: line 5: syntax error near unexpected token `('
./firmware_check.py: line 5: `def fw_check():'
```

cfg.h has:
#define USE_Wire                                           1
#define USE_LiquidCrystal_I2C                       1

- Arduino board: Mega 2560
- Additional hardware:  Sunfounder L2C 2004a LCD screen

Desperandum, 2019-03-26:
I found IDLE IDE was stopping firmware_check from working, which now works under a different IDE; however, it seems that LiquidCrystal_I2C will not initialize in cfg.h. I've attempted changing cfg_check and cfg_defaults to include the libraries as well but have had no luck with that. Anyone have experience with their cfg not updating and having suggestions? Could it be a path problem in one of the files?

Also ran dump.py which returns no digitalPinToPort, which is likely caused by a lack of LiquidCrystal_I2C.

olixr, 2019-10-27:
Unfortunately it can be hard to pin point but I feel I experienced an issue like this when I was trying to get DHT library working. Although I was using the esp8266 and there was a check that was overriding the DHT in my case I believe. 

However I have had a board or two in the past I uploaded the sketch too and the `firmware_check` didn't show me all the libraries I tried to enable. I can't say the exact solution but with changing values between the config files and uploading a few times eventually it seems to have had the libraries enabled.