nanpy/nanpy #110: Compilation Error in Arduino 2.1.0.5

Aprextext, 2019-04-26:
I have followed the installation instructions, but when I try and upload it/compile it to either of my boards (Arduino Uno or Mega 2560), it won't compile, and I get a long, long list of issues. Sadly, I don't really understand any of them!

Nanpy/freeram.c.o: In function `free_ram3':
/home/pi/sketchbook/libraries/Nanpy/freeram.c:53: multiple definition of `free_ram1'
freeram.c.o:freeram.c:53: first defined here
Nanpy/freeram.c.o: In function `free_ram3':
/home/pi/sketchbook/libraries/Nanpy/freeram.c:53: multiple definition of `free_ram2'
freeram.c.o:freeram.c:53: first defined here
Nanpy/freeram.c.o: In function `free_ram3':
/home/pi/sketchbook/libraries/Nanpy/freeram.c:53: multiple definition of `free_ram3'
freeram.c.o:freeram.c:53: first defined here
Nanpy/InfoClass.cpp.o: In function `nanpy::InfoClass::elaborate(nanpy::MethodDescriptor*)':
/home/pi/sketchbook/libraries/Nanpy/InfoClass.cpp:14: multiple definition of `nanpy::InfoClass::get_firmware_id()'
InfoClass.cpp.o:InfoClass.cpp:14: first defined here
Nanpy/InfoClass.cpp.o: In function `nanpy::InfoClass::elaborate(nanpy::MethodDescriptor*)':
/home/pi/sketchbook/libraries/Nanpy/InfoClass.cpp:14: multiple definition of `nanpy::InfoClass::elaborate(nanpy::MethodDescriptor*)'
InfoClass.cpp.o:InfoClass.cpp:14: first defined here
Nanpy/BaseClass.cpp.o: In function `_GLOBAL__sub_D__ZN5nanpy8Register7classesE':
/home/pi/sketchbook/libraries/Nanpy/SlimArray.h:86: multiple definition of `nanpy::Register::classes'
BaseClass.cpp.o:SlimArray.h:86: first defined here
Nanpy/MethodDescriptor.cpp.o: In function `nanpy::MethodDescriptor::returns(unsigned long)':
/home/pi/sketchbook/libraries/Nanpy/MethodDescriptor.cpp:126: multiple definition of `nanpy::MethodDescriptor::MethodDescriptor()'
MethodDescriptor.cpp.o:MethodDescriptor.cpp:126: first defined here
Nanpy/MethodDescriptor.cpp.o: In function `nanpy::MethodDescriptor::returns(unsigned long)':
/home/pi/sketchbook/libraries/Nanpy/MethodDescriptor.cpp:126: multiple definition of `nanpy::MethodDescriptor::stack_pool_size'
MethodDescriptor.cpp.o:MethodDescriptor.cpp:126: first defined here
Nanpy/MethodDescriptor.cpp.o: In function `nanpy::MethodDescriptor::returns(unsigned long)':
/home/pi/sketchbook/libraries/Nanpy/MethodDescriptor.cpp:126: multiple definition of `nanpy::MethodDescriptor::stack_pool'
MethodDescriptor.cpp.o:MethodDescriptor.cpp:126: first defined here
Nanpy/MethodDescriptor.cpp.o: In function `nanpy::MethodDescriptor::returns(unsigned long)':
/home/pi/sketchbook/libraries/Nanpy/MethodDescriptor.cpp:126: multiple definition of `nanpy::MethodDescriptor::MethodDescriptor()'
MethodDescriptor.cpp.o:MethodDescriptor.cpp:126: first defined here
Nanpy/MethodDescriptor.cpp.o: In function `nanpy::MethodDescriptor::returns(unsigned long)':
/home/pi/sketchbook/libraries/Nanpy/MethodDescriptor.cpp:126: multiple definition of `nanpy::MethodDescriptor::getNArgs()'
MethodDescriptor.cpp.o:MethodDescriptor.cpp:126: first defined here
Nanpy/MethodDescriptor.cpp.o: In function `nanpy::MethodDescriptor::returns(unsigned long)':
/home/pi/sketchbook/libraries/Nanpy/MethodDescriptor.cpp:126: multiple definition of `nanpy::MethodDescriptor::getBool(int)'
MethodDescriptor.cpp.o:MethodDescriptor.cpp:126: first defined here
Nanpy/MethodDescriptor.cpp.o: In function `nanpy::MethodDescriptor::returns(unsigned long)':
/home/pi/sketchbook/libraries/Nanpy/MethodDescriptor.cpp:126: multiple definition of `nanpy::MethodDescriptor::getInt(int)'
MethodDescriptor.cpp.o:MethodDescriptor.cpp:126: first defined here
Nanpy/MethodDescriptor.cpp.o: In function `nanpy::MethodDescriptor::returns(unsigned long)':
/home/pi/sketchbook/libraries/Nanpy/MethodDescriptor.cpp:126: multiple definition of `nanpy::MethodDescriptor::getByte(int)'
MethodDescriptor.cpp.o:MethodDescriptor.cpp:126: first defined here
Nanpy/MethodDescriptor.cpp.o: In function `nanpy::MethodDescriptor::returns(unsigned long)':
/home/pi/sketchbook/libraries/Nanpy/MethodDescriptor.cpp:126: multiple definition of `nanpy::MethodDescriptor::getFloat(int)'
MethodDescriptor.cpp.o:MethodDescriptor.cpp:126: first defined here
Nanpy/MethodDescriptor.cpp.o: In function `nanpy::MethodDescriptor::returns(unsigned long)':
/home/pi/sketchbook/libraries/Nanpy/MethodDescriptor.cpp:126: multiple definition of `nanpy::MethodDescriptor::getDouble(int)'
MethodDescriptor.cpp.o:MethodDescriptor.cpp:126: first defined here
Nanpy/MethodDescriptor.cpp.o: In function `nanpy::MethodDescriptor::returns(unsigned long)':
/home/pi/sketchbook/libraries/Nanpy/MethodDescriptor.cpp:126: multiple definition of `nanpy::MethodDescriptor::getString(int)'
MethodDescriptor.cpp.o:MethodDescriptor.cpp:126: first defined here
Nanpy/MethodDescriptor.cpp.o: In function `nanpy::MethodDescriptor::returns(unsigned long)':
/home/pi/sketchbook/libraries/Nanpy/MethodDescriptor.cpp:126: multiple definition of `nanpy::MethodDescriptor::getClass()'
MethodDescriptor.cpp.o:MethodDescriptor.cpp:126: first defined here
Nanpy/MethodDescriptor.cpp.o: In function `nanpy::MethodDescriptor::returns(unsigned long)':
/home/pi/sketchbook/libraries/Nanpy/MethodDescriptor.cpp:126: multiple definition of `nanpy::MethodDescriptor::getObjectId()'
MethodDescriptor.cpp.o:MethodDescriptor.cpp:126: first defined here
Nanpy/MethodDescriptor.cpp.o: In function `nanpy::MethodDescriptor::returns(unsigned long)':
/home/pi/sketchbook/libraries/Nanpy/MethodDescriptor.cpp:126: multiple definition of `nanpy::MethodDescriptor::getName()'
MethodDescriptor.cpp.o:MethodDescriptor.cpp:126: first defined here
Nanpy/MethodDescriptor.cpp.o: In function `nanpy::MethodDescriptor::returns(unsigned long)':
/home/pi/sketchbook/libraries/Nanpy/MethodDescriptor.cpp:126: multiple definition of `nanpy::MethodDescriptor::returns(String&)'
MethodDescriptor.cpp.o:MethodDescriptor.cpp:126: first defined here
Nanpy/MethodDescriptor.cpp.o: In function `nanpy::MethodDescriptor::returns(unsigned long)':
/home/pi/sketchbook/libraries/Nanpy/MethodDescriptor.cpp:126: multiple definition of `nanpy::MethodDescriptor::returns(char const*)'
MethodDescriptor.cpp.o:MethodDescriptor.cpp:126: first defined here
Nanpy/MethodDescriptor.cpp.o: In function `nanpy::MethodDescriptor::returns(unsigned long)':
/home/pi/sketchbook/libraries/Nanpy/MethodDescriptor.cpp:126: multiple definition of `nanpy::MethodDescriptor::returns(char)'
[... 72 lines cut ...]
ComChannel.cpp.o:ComChannel.cpp:158: first defined here
Nanpy/ColorSensorClass.cpp.o: In function `nanpy::ColorSensorClass::elaborate(nanpy::MethodDescriptor*)':
/home/pi/sketchbook/libraries/Nanpy/ColorSensorClass.cpp:11: multiple definition of `nanpy::ColorSensorClass::get_firmware_id()'
ColorSensorClass.cpp.o:ColorSensorClass.cpp:11: first defined here
Nanpy/ColorSensorClass.cpp.o: In function `nanpy::ColorSensorClass::elaborate(nanpy::MethodDescriptor*)':
/home/pi/sketchbook/libraries/Nanpy/ColorSensorClass.cpp:11: multiple definition of `nanpy::ColorSensorClass::elaborate(nanpy::MethodDescriptor*)'
ColorSensorClass.cpp.o:ColorSensorClass.cpp:11: first defined here
Nanpy/WatchdogClass.cpp.o: In function `disable_watchdog_at_startup()':
/home/pi/sketchbook/libraries/Nanpy/WatchdogClass.cpp:37: multiple definition of `disable_watchdog_at_startup()'
WatchdogClass.cpp.o:WatchdogClass.cpp:37: first defined here
Nanpy/DefineClass.cpp.o: In function `nanpy::DefineClass::elaborate(nanpy::MethodDescriptor*)':
/home/pi/sketchbook/libraries/Nanpy/DefineClass.cpp:67: multiple definition of `nanpy::DefineClass::get_firmware_id()'
DefineClass.cpp.o:DefineClass.cpp:67: first defined here
Nanpy/DefineClass.cpp.o: In function `nanpy::DefineClass::elaborate(nanpy::MethodDescriptor*)':
/home/pi/sketchbook/libraries/Nanpy/DefineClass.cpp:67: multiple definition of `nanpy::DefineClass::elaborate(nanpy::MethodDescriptor*)'
DefineClass.cpp.o:DefineClass.cpp:67: first defined here
Nanpy/RAMClass.cpp.o: In function `nanpy::RAMClass::elaborate(nanpy::MethodDescriptor*)':
/home/pi/sketchbook/libraries/Nanpy/RAMClass.cpp:15: multiple definition of `nanpy::RAMClass::get_firmware_id()'
RAMClass.cpp.o:RAMClass.cpp:15: first defined here
Nanpy/RAMClass.cpp.o: In function `nanpy::RAMClass::elaborate(nanpy::MethodDescriptor*)':
/home/pi/sketchbook/libraries/Nanpy/RAMClass.cpp:15: multiple definition of `nanpy::RAMClass::elaborate(nanpy::MethodDescriptor*)'
RAMClass.cpp.o:RAMClass.cpp:15: first defined here
collect2: error: ld returned 1 exit status

`   `

olixr, 2019-10-27:
At a glance of the logs it looks like you have multiple imports of the library going on. Make sure you only have the one nanpy version installed. 

Check your libraries folder and see if you have some duplicates going on.

apollowebdesigns, 2020-02-20:
@Aprextext it is what @olixr has suggested.

I had exactly the same issue on the same Arduino IDE version.

For anyone else suffering this, do the following below:

- On linux system (in this case Raspbian Buster): 

```
sudo apt-get update
sudo apt-get install arduino
```
- I deleted Nanpy firmware from sketchbook AND sketchbook/libraries
- I am on a linux system so verified the deletes with this with `find / -name Nanpy`
- Then, reinstalled following instructions from their [GitHub page](https://github.com/nanpy/nanpy-firmware)

You should then be able to upload the software from File > Sketchbook > Nanpy inside the Arduino IDE.